zmkfirmware/zmk #10: RFC: Left/right hand support

petejohanson, 2020-06-11:
Before we can implement split transports, we need each side of a left/right hand split keyboard to know which hand they are.

In particular, in order to send the correct *column* values for KSCAN events passed over the split transport, a right hand board needs to translate the column by a certain offset before sending the row/column/pressed state over the transport.

# Proposal

"Left/Right" detection can be implemented as separate DT overlay files that are included for a separate build/flash cycle for each side of the keyboard. Doing so would allow the following:

1. Differences in hardware present on each side of the split can be account for at *build time*, not *runtime*, allowing for preserving space not needed on a given side, and more importantly not trying to access devices over i2c for example, that are not present.
1. Offsets for KSCAN events can be properly set at build time, and don't need to be set via runtime detection.

# Alternative

The left/right hand is set via EEPROM, NVS, or Zephyr settings API, which matches how QMK implements this.

thinkingoutsidethebox42, 2020-06-11:
Sounds good.

Presumably that extends to any extra parts such as keypads and pedals without too much hassle?

I'd suggest trying build time (DT overlays) as the initial implementation attempt because it leverages Zephyr's strengths.  And then evaluate it.  We could always refactor it into a realtime setting if the needs arises.

petejohanson, 2020-06-11:
@thinkingoutsidethebox42 Yeah, that would allow us to do things like define an encoder that's only on one side, or an OLED there, etc.

thinkingoutsidethebox42, 2020-06-11:
Sorry, I meant as first class citizens ... left, right, keypad, pedals?

petejohanson, 2020-06-11:
@thinkingoutsidethebox42  Yeah, we could also use this for keypads, if they are truly part of the "composite device", and not just their own standalone HID device that connects to a host

thinkingoutsidethebox42, 2020-06-11:
Cool.  And presumably switch modes to act as either within the composite device or as a seperate HID if connected directly to the PC/mobile/tablet.  Sounds great!

innovaker, 2020-06-24:
Has the thinking on this changed much given the recent big picture discussions on Discord and ongoing refactors?

closed 2020-08-11 by petejohanson