zmkfirmware/zmk #11: FAQs (initial version)

thinkingoutsidethebox42, 2020-06-11:
Initiail version of FAQs.  Required by #7.

Currently a realtime work-in-progress on Google Docs.  Please contact me on Discord if you need the link to contribute.

innovaker, 2020-06-12:
Sidebar location?  Within Basics or between Basics & Development?

petejohanson, 2020-06-12:
I wonder if "Basics" is just wrong? Better top level label, so we could put both Intro and FAQ under there?

innovaker, 2020-06-12:
That crossed my mind this morning.  Agreed.

petejohanson, 2020-06-12:
"Overview"?

innovaker, 2020-06-12:
Perhaps.  When working with unfamiliar tech, I always look for "Introduction" and "Getting Started" as key words.

petejohanson, 2020-06-12:
I vote for the group being renamed to "Getting Started", which means we can leave the first "Introduction" page as is, and add FAQ somewhere below in the same group.

innovaker, 2020-06-12:
Sounds good!

PR #22 (2020-06-13): docs: FAQs (initial version)
Closes #11.
closed 2020-06-13 by petejohanson