zmkfirmware/zmk #12: Development Basic Setup on Raspberry Pi OS (Buster)

thinkingoutsidethebox42, 2020-06-11:
Good job on the instructions!

I've encountered two issues so far.

## Install Dependencies
Running
```
sudo apt install -y
```
it complained that `gcc-multilib` wasn't available which I bypassed by removing it from the command.

## Install Zephyr Python Dependencies

Running
```
pip3 install --user -r zephyr/scripts/requirements.txt
```
it complained

```
Collecting cmsis-pack-manager>=0.2.7 (from pyocd>=0.24.0->-r zephyr/scripts/requirements-run-test.txt (line 9))
  Downloading https://files.pythonhosted.org/packages/c9/2b/1ac9b85b5ba535ba386467cb9a897e3f460c04f34a305e02b22f86b65367/cmsis-pack-manager-0.2.10.tar.gz (45kB)
    100% |████████████████████████████████| 51kB 2.4MB/s
    Complete output from command python setup.py egg_info:
    Package libffi was not found in the pkg-config search path.
    Perhaps you should add the directory containing `libffi.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'libffi' found
    Package libffi was not found in the pkg-config search path.
    Perhaps you should add the directory containing `libffi.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'libffi' found
    Package libffi was not found in the pkg-config search path.
    Perhaps you should add the directory containing `libffi.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'libffi' found
    Package libffi was not found in the pkg-config search path.
    Perhaps you should add the directory containing `libffi.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'libffi' found
    Package libffi was not found in the pkg-config search path.
    Perhaps you should add the directory containing `libffi.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'libffi' found
    c/_cffi_backend.c:15:10: fatal error: ffi.h: No such file or directory
     #include <ffi.h>
              ^~~~~~~
    compilation terminated.
 ...
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-e44et2su/cmsis-pack-manager/
```

so I installed `libffi-dev`
```
sudo apt-get install libffi-dev
```
and then reran
```
pip3 install --user -r zephyr/scripts/requirements.txt
```
it then complained
```
  Downloading https://files.pythonhosted.org/packages/ae/e7/d9c3a176ca4b02024debf82342dab36efadfc5776f9c8db077e8f6e71821/pycparser-2.20-py2.py3-none-any.whl (112kB)
    100% |████████████████████████████████| 112kB 1.9MB/s
Building wheels for collected packages: cmsis-pack-manager
  Running setup.py bdist_wheel for cmsis-pack-manager ... error
  Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-45ylt83v/cmsis-pack-manager/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-bglvsvky --python-tag cp37:
  running bdist_wheel
  running build
  running build_py
  creating build/lib
  creating build/lib/cmsis_pack_manager
  copying cmsis_pack_manager/__init__.py -> build/lib/cmsis_pack_manager
  copying cmsis_pack_manager/pack_manager.py -> build/lib/cmsis_pack_manager
  copying cmsis_pack_manager/_native__ffi.py -> build/lib/cmsis_pack_manager
  copying cmsis_pack_manager/_version.py -> build/lib/cmsis_pack_manager
  copying cmsis_pack_manager/_native.py -> build/lib/cmsis_pack_manager
  error: [Errno 2] No such file or directory: 'cargo': 'cargo'

  ----------------------------------------
  Failed building wheel for cmsis-pack-manager
```
which I *thought* was due to an older version of pip, according to https://github.com/ARMmbed/cmsis-pack-manager/issues/137
So I upgraded pip:
```
python3 -m pip install --upgrade pip
```
but it didn't help the cmsis/cargo error.

I'm still looking into this.  Any thoughts please?

petejohanson, 2020-06-11:
So... likely this is failing because their upstream does not push prebuilt Python packages for ARM, which means you are triggering the "build it locally" behaviour, which seems to require Rust (Yay Rust!) and it's build tool `cargo`.

If you install the Rust and Cargo packages (you'll have to google their debian names, not sure off the top of my head), then try again, any better?

thinkingoutsidethebox42, 2020-06-11:
Thanks!

Both were installed successfully with 
```
sudo apt install rustc
```

This time ...
```
pip3 install --user -r zephyr/scripts/requirements.txt
```
produced what looks like rustc errors about `alloc`?  Please see the attached log as my hands-on rust experience is currently this activity.
[cmsis-pack-manager.log](https://github.com/zmkfirmware/zmk/files/4767191/cmsis-pack-manager.log)

innovaker, 2020-06-14:
On the advice of @petejohanson, I used
```
pip3 install --user -r zephyr/scripts/requirements-base.txt
```
which worked!

I then added `source zephyr/zephyr-env.sh` to `~/.bashrc` which is slightly different than the instructions, using:
```
echo 'source ~/zmk/zephyr/zephyr-env.sh' >> ~/.bashrc
```

Then
```
cd app
```

I then tried building with:
```
west build -b nrf52840dongle_nrf52840 -- -DKEYMAP=DEFAULT
```
but it complained about the compiler.

So I installed the cross-compiler following
https://docs.zephyrproject.org/latest/getting_started/toolchain_other_x_compilers.html
by installing the `gcc-arm-non-eabi`
```
sudo apt install gcc-arm-none-eabi
```
and by editing `~/.zephyrrc` to:
```
export ZEPHYR_TOOLCHAIN_VARIANT=cross-compile
export CROSS_COMPILE=/usr/bin/arm-none-eabi-
export ZEPHYR_SDK_INSTALL_DIR=/home/pi/.local/zephyr-sdk-0.11.2
```

I tried the build again and it worked ... until it hit a compile error related to keymaps (because there isn't one for `nrf52840dongle_nrf52840`.

But an established board and shield
```
west build -b proton_c -- -DSHIELD=petejohanson_proton_handwire
```
built successfully!

Hopefully my experiences will help us improve the build guide for cross-platform support.

When I get more time I'll look into trying a basic keymap matrix and flashing the nRF52840 dongle!

Note for the future:  pip3 gave deprecation warnings so we should probably look into it.

PR #55 (2020-07-14): Multiple OS tabs for dev setup instructions
Closes #12 and #26
closed 2020-07-14 by petejohanson