zmkfirmware/zmk #21: Standardize HID constants

innovaker, 2020-06-13:
- Revisit/refactor current `keys.h`
- Include any useful values for USB or HOGP from HID Usage Tables v1.12:
https://www.usb.org/sites/default/files/documents/hut1_12v2.pdf
- Consider values and naming conventions found in existing KB firmware, Linux, Windows, Android, etc.
- Decide upon any conventions for ZMK.
- Open up consultation with the ZMK contributor community before finalizing.

petejohanson, 2020-08-08:
Updated usage tables version: https://usb.org/sites/default/files/hut1_2.pdf

tominabox1, 2020-09-04:
Add scope of #158 to this

innovaker, 2020-09-09:
Let's make a concerted effort to nail this down, including @tominabox1's efforts on #161 and taking into account the effects of #86.

innovaker, 2020-09-21:
It's RFC time!  Feedback, suggestions and discussions welcomed!  Everything is still on the table at this point.

To facilitate this first pass, we're using Google Sheets as a _living document_.  It contains the proposed names (tags) and descriptions that are used for the `key-press` behavior (`&kp`, `&cp`):

RFC: https://docs.google.com/spreadsheets/d/1s6URDIL7pqcMiUvY8s0esWvLIA1dKNmHqp6jEUtrqTw/

- Feel free to make _focused comments_ (per cell) in the sheet itself, but please use this GitHub issue for more general or _cross-cutting_ discussions.
- Please **_let people know who you are_**, *either* by:
  - signing into Google with an account that has the same visible name as your Discord/GitHub name, OR ...
  - adding your Discord/GitHub name onto each comment (i.e. `-innovaker`).
- The _white cells_ contain the stuff that will be put into ZMK.  This is the focus of the RFC.
- The _grey cells_ contain useful background information from other sources.  Please don't comment on these cells.
- I'll be updating the document as each thread of discussion progresses, so it will change over time.
- Be sure to login to the sheet if you want to receive ongoing notifications to comments.
- Care was taken to ensure consistency, but there's bound to be some oversights.  So ... what can you spot!? :)

**Design Considerations (thus far)**
- There are many ways to slice the "keys cake" and everyone tends to have their own opinions/biases.  Taking this into consideration, I'm suggesting we support a variety of logical aliases for each code - both long and short forms.
- Due to C's namespace limitations and overlaps, the prefix naming conventions _at time of writing this post_ are: 
   - Keyboard: _none_
   - Keypad: `KP_`
   - Consumer: `CSM_` 
- I haven't included the entire consumer page as we don't yet have the capacity for that scale of experimenting or testing.  I filtered it down to those supported within existing smartphones or keyboards, plus a few more I thought might be useful.
- I've kept existing key codes unless it was there was a better alternative.  So there may be a few breaking changes.
- Support for each code depends on the operating system.

innovaker, 2020-09-21:
For those wondering what the code might look like, here's a preview:
```
/* Keyboard Escape */
#define ESCAPE (HID_USAGE_KEY_KEYBOARD_ESCAPE)
#define ESC    (HID_USAGE_KEY_KEYBOARD_ESCAPE)

/* Keyboard Space */
#define SPACE (HID_USAGE_KEY_KEYBOARD_SPACEBAR)
#define SPC   (HID_USAGE_KEY_KEYBOARD_SPACEBAR)
```
We'll RFC the code itself as a PR after the initial sheets RFC is completed.  Please hold off comments on style until then.
Note: The `HID_USAGE_` variables - effectively HID Usage Table definitions - will soon appear as a prerequisite PR.

innovaker, 2020-09-21:
I've a couple of general questions to kick off ...

1. Namespace pollution aside, can anyone foresee any potential issues with multiple aliases?

2. Should key names which have two words (e.g. `VOLUME UP`) be joined with an underscore (`VOLUME_UP`) or _no gap_ (`VOLUMEUP`)?

3. Likewise to (2), should prefixes (e.g. `KP_`) use an underscore or _no gap_?

innovaker, 2020-09-21:
Regarding interoperability support for other firmware (i.e. QMK, BlueMicro), I'm considering that a separate issue because it can be handled different ways, such as additional header files or key map conversion scripts.

innovaker, 2020-09-21:
I'm particularly interested in suggestions for the consumer keys.
They currently follow the HID Usage Table conventions:
`CSM_AL_`: Application Launch
`CSM_AC_`: Application Control
because otherwise these namespaces conflict with the default consumer namespace `CSM_`.  e.g. `CSM_STOP` vs. `CSM_AC_STOP`.
I'd welcome a smarter, more intuitive approach though.

tominabox1, 2020-09-21:
> I've a couple of general questions to kick off ...
>
> 1. Namespace pollution aside, can anyone foresee any potential issues with multiple aliases?
> 2. Should key names which have two words (e.g. [...]

1. I don't see the value of multiple aliases personally, but I don't see any reason it is a problem from my prospective anyway.
2. I think I would prefer _
3. Users coming from elsewhere will be accustomed to KP_, I would be inclined to offer this.

> I'm particularly interested in suggestions for the consumer keys.
> They currently follow the HID Usage Table conventions:
> `CSM_AL_`: Application Launch
> `CSM_AC_`: Application Control
> because otherwise these namespaces conflict with the default consumer namespace `CSM_`. e.g.: `CSM_STOP` ≠ `CSM_AC_STOP` ???
> I'd welcome a smarter, more intuitive approach though. Perhaps by using the `MEDIA` convention? I'll look into it further.

Do these Consumer keys even do anything on modern systems?

innovaker, 2020-09-21:
Thanks!

> Do these Consumer keys even do anything on modern systems?
Honestly I'm not 100% without checking them on all OS's.  The vast majority are documented on Android though.

innovaker, 2020-09-21:
As an experiment, I dropped the `CSM_` namespace from the entire consumer page and compared it against the keyboard/keypad namespace.  These were the conflicts:
```
Keyboard/Keypad: Keyboard Menu
Consumer: Menu

Keyboard/Keypad: Keyboard Help
Consumer: Help

Keyboard/Keypad: Keyboard Power
Consumer: Power

Consumer: Sleep Mode
Consumer: Sleep

Keyboard/Keypad: Keyboard Mute
Consumer: Mute

Keyboard/Keypad: Keyboard Pause
Consumer: Pause

Keyboard/Keypad: Keyboard Stop
Consumer: Stop

Keyboard/Keypad: Keyboard Volume Down
Consumer: Volume Decrement

Keyboard/Keypad: Keyboard Volume Up
Consumer: Volume Increment
```

So it might actually be possible to drop the CSM prefix entirely if we use `POWER_` and `MEDIA_` conventions (or `P_` and `M_`) for the last 7 on that list.  That only leaves `MENU` and `HELP` as conflicts???

It opens the door to potential conflicts in the future, because the consumer page is still expanding.  But I'd suggest the risk is minimal.

Any thoughts?

innovaker, 2020-09-21:
That said, there could also be conflicts with the _unofficial_ codes that Android supports on the keyboard page if we chose to support those too.  Those are at the very bottom of the sheet.

innovaker, 2020-09-21:
As a second, independent experiment, this time I dropped the `AL_` and `AC_` conventions (inherited from the HID specification).  These were the conflicts:
```
Consumer AC Cancel
Keyboard Cancel

Consumer AC Copy
Keyboard Copy

Keyboard Cut
Consumer AC Cut

Consumer Quit
Consumer AC Exit

Keyboard Find
Consumer AC Find

Consumer Help
Consumer AL Integrated Help Center

Consumer AL Terminal Lock/Screensaver
Consumer AC Lock

Keyboard Menu
Consumer Menu

Keyboard Mute
Consumer Mute

Keyboard Paste
Consumer AC Paste

Keyboard Pause/Break
Consumer Pause

Keyboard Power
Consumer Power

Keyboard Again
Consumer AC Redo/Repeat

Consumer Sleep Mode
Consumer Sleep

Keyboard Stop
Consumer Stop
Consumer AC Stop

Keyboard Undo
Consumer AC Undo

Keyboard Volume Down
Consumer Volume Decrement

Consumer Volume Increment
Keyboard Volume Up

Consumer Media Select WWW
Consumer AL Internet Browser

Keyboard Page Down
Keyboard Page Down (Linux/Android?)

Keyboard Stop CD (Linux/Android?)
Keyboard Stop (Linux/Android?)
```
Hopefully these experiments illustrate the complexities of finding a neat solution to the namespace dilemma.

innovaker, 2020-09-21:
I'm currently leaning on replacing `CSM_` with `C_` but keeping `AL_` and `AC_` until someone comes up with a smarter suggestion.

okke-formsma, 2020-09-22:
Do we have information on how well supported all of these codepages and keycodes are on the various platforms?  I think it makes sense to keep the prefixes on anything that's not widely supported and/or crossplatform.

innovaker, 2020-09-22:
Everything listed so far is based on research (please see the sources sheet).  It's tricky because we rely on documentation.  But my understanding is that keyboard/keypad page is generally universal.  The consumer page - the ones I've shortlisted - depend on the platform.  But they're listed in Android (and Linux), which covers the majority of non-Windows devices.  As for iOS, I didn't find much but I'm not part of their developer network.

innovaker, 2020-09-22:
To follow that up, I'm *considering* everything on the keyboard/keypad and consumer pages because you have to look at the big picture to identify workable (hopefully future-proof) naming conventions.  But I've *shortlisted* the consumer page codes to those documented by existing platforms (that's what's on the RFC sheet).

In terms of testing, we don't currently have the capacity to methodically test the codes across all the various platforms.  So for now, I'm going by available documentation on the basis that we'll test/fix and find cross-platform issues later.

That said, perhaps we shouldn't *document* the untested key codes until we empirically compare the platforms?

innovaker, 2020-09-22:
I need some help please!  You'll need a working ZMK build.

Please could you test the differences within these sets of codes with your OS?  I expect them to work on Linux, maybe Android.  I'm not sure about Windows/iOS.

**Set 1**
These should be page/scroll *down* codes, so you can probably use a long webpage.  But there might be subtle differences in the scrolling style?  I'm not sure.
`&kp 0x4E`
`&kp 0xF6`

**Set 2**
These should be page/scroll *up* codes, so you can probably use a long webpage.  But there might be subtle differences in the scrolling style?  I'm not sure.
`&kp 0x4B`
`&kp 0xF5`

**Set 3**
These should be *stop* codes.  But they might be "stop browser/activity" or "stop media".  So I'd suggest playing a song on the OS's default music player, and then test the key whilst trying to reload a webpage (i.e. stop).  If you know of any application that also accepts a stop command, that would be a useful test too.
`&kp 0x78`
`&kp 0xE9`
`&kp 0xF3`

Thanks!

innovaker, 2020-09-22:
Aside from the above, I've mostly confirmed that the **unofficial** *keyboard* keys (Linux/Android?) are duplicates of HID **official** *keyboard* keys.  Weird.  Perhaps there's OS nuances that we'll encounter later.  For now I suggest we just number them, until we understand them better:
e.g. `MUTE` and `MUTE2`, `FIND` and `FIND2`, `BACK` and `BACK2`, etc.

innovaker, 2020-09-22:
At time of writing, the following codes are set to be relegated to legacy support in favour of the proposed aliases.  You'll still be able to use the legacy codes after this issue is merged, but legacy codes will likely be removed in a later version of ZMK.  If this matters to you, I suggest checking the relevant cell on the RFC sheet and leaving a comment on that cell if you'd like it to be reconsidered.  Thanks!
```
NUM_1
NUM_2
NUM_3
NUM_4
NUM_5
NUM_6
NUM_7
NUM_8
NUM_9
NUM_0
TILD
QUOT
SCLK
PAUS
RARW
LARW
DARW
UARW
KDIV
KMLT
KMIN
KPLS
LPRN
RPRN
LCUR
RCUR
CRRT
PRCT
LABT
RABT
AMPS
PIPE
COLN
KSPC
ATSN
BANG
M_NEXT
M_PREV
M_STOP
M_EJCT
M_PLAY
M_MUTE
M_VOLU
M_VOLD
```

innovaker, 2020-09-23:
I've added a new column to the RFC sheet called `Legacy`.  This shows which existing ZMK codes would be relegated to legacy status.  The reasoning depends on each code, but its usually due to:
* being replaced by a convention (i.e. `KP_`)
* being superseeded by better alternatives
* because it requires a modifier (see #86)

Legacy codes will still be usable after the merge of this issue, but would not be documented and will eventually be phased out in the future.  So if one of the relegations bothers you, please leave a comment on the appropriate row.

innovaker, 2020-09-24:
Until someone has a better idea, I've applied the numeric suffixes as described in: https://github.com/zmkfirmware/zmk/issues/21#issuecomment-696704156
for the keys that don't need further identification described by: https://github.com/zmkfirmware/zmk/issues/21#issuecomment-696700763

petejohanson, 2020-09-24:
One comment: I think it's *good* to prefix the consumer page stuff. We should keep those "clean" and then consider helpful aliases/shorthands from there for a subset.

innovaker, 2020-09-24:
@petejohanson, have you any thoughts on the strategy for introducing/testing the new codes please?  I've previously suggested that we throw them out there and see what sticks (because it's based on documentation they *should* all just depend on the OS), but that does leave open the *possibility* of *potentially* having to pull some codes ... for whatever reason once testing is done.  It's akin to throwing out the alpha/beta to everyone vs. drip feeding after thorough testing.

petejohanson, 2020-09-24:
This really does need user testing on a variety of OS's to really understand what works/doesn't. I would suggest we move forwards with this, and then test, and refactor as needed. I think as long as the "core" still works, then minor tweaks/breakage for consumer keys, etc we can handle and fix.

innovaker, 2020-10-01:
Following up my earlier request: https://github.com/zmkfirmware/zmk/issues/21#issuecomment-696700763

I've tested:
- `0x4E` vs `0xF6`
- `0x4B` vs `0xF5`
on Windows, Linux and Android.  iOS wasn't tested.

Support:
- `0x4E` and `0x4B` worked on all platforms.
- `0xF6` and `0xF5` only worked on Linux/Android.

Linux (`evtest`) registered the events as:
- `0x4E`: `KEY_PAGEDOWN`
- `0x4B`: `KEY_PAGEUP`
- `0xF6`: `KEY_SCROLLDOWN`
- `0xF5`: `KEY_SCROLLUP`

So I've named the ZMK aliases following this convention.

innovaker, 2020-10-01:
Following up my earlier request: #21 (comment)

I've tested:
- `&cp 0xB7`
- `&cp 0xCC`
- `&cp 0x226`
- `&kp 0x78`
- `&kp 0xE9`
- `&kp 0xF3`

on Windows, Linux and Android.  iOS wasn't tested.

### Windows
Tested using VLC, Windows Media Player, Firefox and Chrome:
- `&cp 0xB7`: Stops VLC.  Stops WMP.
- `&cp 0xCC`: No response.
- `&cp 0x226`: *Pauses* VLC.  No response from WMP.  Triggers stop in Chrome and Firefox.
- `&kp 0x78`: No response.
- `&kp 0xE9`: No response.
- `&kp 0xF3`: No response.

### Linux
Tested with `evtest`:
- `&cp 0xB7`: `KEY_STOPCD`
- `&cp 0xCC`: `KEY_UNKNOWN`
- `&cp 0x226`: `KEY_STOP`
- `&kp 0x78`: `KEY_STOP`
- `&kp 0xE9`: `KEY_STOPCD`
- `&kp 0xF3`: `KEY_STOP`

### Android
Tested with Google Music, Chrome and Firefox:
- `&cp 0xB7`:  Stops music.  No effect on Chrome or Firefox.
- `&cp 0xCC`: No response.  No effect on Chrome or Firefox.
- `&cp 0x226`: Stops music.  No effect on Chrome or Firefox.
- `&kp 0x78`: Stops music.  No effect on Chrome or Firefox.
- `&kp 0xE9`: Stops music.  No effect on Chrome or Firefox.
- `&kp 0xF3`: Stops music.  No effect on Chrome or Firefox.

innovaker, 2020-10-01:
Based on the above results, I've gone with ...
- `0x0007 0x0078`: `STOP`
- `0x0007 0x00E9`: `STOP2`
- `0x0007 0x00F3`: `STOP3`
- `0x000C 0x00B7`: `C_STOP`
- `0x000C 0x00CC`: `C_STOP_EJECT`
- `0x000C 0x0226`: `C_AC_STOP`

which means most users will want to use `&cp C_STOP` for media/music controls, and `&cp C_AC_STOP` for internet browsers.  

This works well, because the other media controls seem to be (*not tested*) `C_PLAY`, `C_PAUSE_PLAY`, `C_PAUSE`, `C_RECORD`, `C_NEXT`, `C_PREVIOUS`, `C_STOP`, `C_EJECT` and `C_REPEAT`, `C_MUTE`, `C_VOLUME_UP` and `C_VOLUME_DOWN`.  For anyone wondering, the current `M_` aliases point to some of these codes.  The `M_` prefix is *currently* destined to be *deprecated*.

The keyboard (`&kp`) `STOP, STOP2, STOP3` keys don't seem to be so widely supported and frankly I don't know the true intended meaning behind them beyond the description `Keyboard Stop`.

This decision brings all outstanding conflicts to their resolution.

innovaker, 2020-10-11:
Whilst working through the documentation (#218), this occurred to me regarding the *keyboard* page `0x07`.

- codes <= `0x64`, and modifiers:
   - are boot-capable.
   - it's standard across all platforms - PC, Mac, Linux.
- *some* codes > `0x64` (often *functional* keys):
   - seem to have mixed support across platforms.
   - codes such as `STOP` aren't well defined!
   - some overlap with the consumer page, such as:
     - `VOLUME_UP`, `VOLUME_DOWN`
     - `CUT`, `COPY`, `PASTE`
     - `UNDO`, `REDO`
     - `PLAY_PAUSE`, `NEXT`, `PREV`, `STOP`, etc.
- We'll probably find some codes are platform specific.
- We probably won't know what's best for each platform until we've done thorough testing, which will take a while.
- We may need *virtual keys* for device profiles.  For instance, the same key name `CUT` might theoretically:
  - send `CTRL`+`C` connected to your PC
  - send `C_AC_CUT` connected to your Mac
  - send `(keyboard) CUT` to your smartphone
- *Virtual keys* might also be important for Keyboard Layouts/Internationalisation (#177).
- If we get it wrong from the outset, it could get very messy.  Codes like `CUT` would be a nightmare to deprecate.

### Solution?
So that *this current issue* can move forward in safety, I'm going to add a `K_` prefix to anything that:
 - has overlap with the consumer page, or
 - an unclear purpose, or 
 - uncertain platform support.

We can then decide prefix-less codes once the system has matured and OS testing is complete.

This would include ...
```
0x0007 0x0066	Keyboard Power
0x0007 0x0074	Keyboard Execute
0x0007 0x0075	Keyboard Help
0x0007 0x0076	Keyboard Menu
0x0007 0x0077	Keyboard Select
0x0007 0x0078	Keyboard Stop
0x0007 0x0079	Keyboard Again
0x0007 0x007A	Keyboard Undo
0x0007 0x007B	Keyboard Cut
0x0007 0x007C	Keyboard Copy
0x0007 0x007D	Keyboard Paste
0x0007 0x007E	Keyboard Find
0x0007 0x007F	Keyboard Mute
0x0007 0x0080	Keyboard Volume Up
0x0007 0x0081	Keyboard Volume Down
0x0007 0x009B	Keyboard Cancel
0x0007 0x00E8	Keyboard Play/Pause
0x0007 0x00E9	Keyboard Stop
0x0007 0x00EA	Keyboard Previous
0x0007 0x00EB	Keyboard Next
0x0007 0x00EC	Keyboard Eject
0x0007 0x00ED	Keyboard Volume Up
0x0007 0x00EE	Keyboard Volume Down
0x0007 0x00EF	Keyboard Mute
0x0007 0x00F0	Keyboard WWW
0x0007 0x00F1	Keyboard Back
0x0007 0x00F2	Keyboard Forward
0x0007 0x00F3	Keyboard Stop
0x0007 0x00F4	Keyboard Find
0x0007 0x00F5	Keyboard Scroll Up
0x0007 0x00F6	Keyboard Scroll Down
0x0007 0x00F7	Keyboard Edit
0x0007 0x00F8	Keyboard Sleep
0x0007 0x00F9	Keyboard Lock
0x0007 0x00FA	Keyboard Refresh
0x0007 0x00FB	Keyboard Calculator
```

It's not a perfect approach, but I think it's better for the future.

Any thoughts welcomed.

okke-formsma, 2020-10-16:
As discussed on Discord, I think we should try to limit ourselves to one or two names for a specific keycode. For example, there are six writings for slash in the file:

* SLASH
* SLSH
* SLH
* FSLASH
* FSLSH
* FSLH

I think it's important to have preferably one way to write a keycode. In some cases a short (8 chars or less) and verbose version would be in order, but that seems to be enough for all purposes.

I don't have strong opinions on which codes are picked as long as we make a choice and stick to it. In the case of slash, I think just `SLASH` would be perfectly fine. Short versions are not required for keys outside of the default keyboard/keypad HID, as they are used less often. People can always define their own short versions for a few keys if they care about these things.

innovaker, 2020-10-16:
Thanks for the input @okke-formsma.  That sounds along the same lines that we have in mind.

I've prepared the RFC sheet for @petejohanson to review for the purpose when he gets a chance.

innovaker, 2020-10-16:
Just to add to that for anyone watching who wants to be involved ... please add comments on specific keys to the RFC sheet so that we can track per item comments in a single location! 👍

PR #236 (2020-11-04): feat: HID Usages + Keys Standardisation
- Adds HID Usages for ZMK (#217).
- Standardizes key aliases/codes (#21 - still open for RFC).
- Enhances HID reports to take advantage of the new codes (#234).

### Issues
- Closes #217.
- Closes #21.
- Closes #234.
- Supersedes and closes #246.
- Supersedes and closes #158.
- Supersedes and closes #161.

### Testing
- End-to-end tests covering:
  - all key aliases
  - both the keyboard (keypad) and consumer HID reports
- ZMK on Nordic's nRF52840 Dongle.
  - `kscan-mock` running on dongle systematically pressed and released each key.
- Node scripts on R [... cut]
commit a65295c2 (2020-11-04): feat(keys): Standardize keys

Deprecate superseded names with comments.

Add standardized key names.

Closes #21.  Closes #158.  Closes #161.  Closes #246.
closed 2020-11-04 by petejohanson