zmkfirmware/zmk #34: User-friendly Build Facade/Script

innovaker, 2020-06-24:
- A CLI facade/script that serves as the main point of entry for most users.
- Wraps up west and other tooling.
  - Similar to how west wraps around git, cmake, ninja, etc.
- Does sanity checks on parameters.
  - e.g. checks the board, shield, keymap etc.
- Provides rudimentary error handling.
  - Provides useful feedback for common errors.
  - Points users to ZMK help for unexpected errors.
- Perhaps provides an interactive prompt/menu for commands and parameters.
- Perhaps caches parmeters.
- Probably written in Python as this is the scripting language most likely to be available.

innovaker, 2020-06-24:
Should build upon multiplatform exploration started by #12 and #26.

innovaker, 2020-09-23:
@petejohanson, would we still like to proceed with this idea - not as a replacement for `setup.sh` - but as a development tool?

If so, then let's add it to Hacktoberfest.

petejohanson, 2020-09-23:
I'm kind of on the fence here. To a certain degree, this "hides" west, but only so well. I would rather consider what we really want to get from such a tool before we suggest it fora newcomer. Having someone spend a bunch of time on something, that we then say "Nah, that's not what we want!" really would stink.

okke-formsma, 2020-11-14:
I have set up one `west` command (`west test`) as a proof of concept; https://github.com/zmkfirmware/zmk/pull/385

caksoylar, 2023-04-20:
For reference, @joelspadin developed https://github.com/joelspadin/zmk-setup-script that currently encapsulates the user setup script in a Python-based CLI that uses an extensible `zmk` module.

joelspadin, 2023-04-20:
After using QMK again recently, I was thinking of borrowing some ideas from that, such as having a command such as `zmk install` which creates a config repo and remembers where it is, and then all further commands run from anywhere will operate on that repo unless otherwise specified.

That setup script might then be split across `zmk install` and `zmk add-keyboard` commands, with `zmk` being a script you can pip install.

joelspadin, 2024-08-18:
https://github.com/joelspadin/zmk-cli is an attempt to do this. It currently handles
* Walking you through creating a config repo and cloning it to your PC
* Copying stock keymaps for keyboards into the config repo
* Opening keymaps in a text editor
* Adding/removing keyboards from the GitHub build
* Opening a browser to the GitHub actions page
* Adding/removing Zephyr modules
* Creating new boards/shields from a template (limited number of templates currently, but we can expand this later)

It does not currently handle flashing keyboards, but that could potentially be implemented. 

It intentionally does not handle building the firmware or setting up a dev environment. I currently consider that out of scope for what this tool is intended to do, but if enough people want that, we could try adding it.

Townk, 2024-08-18:
I wish I knew about this project before I created my [zmk-cli](https://github.com/Townk/zmk-cli). Although yours and mine are dealing with different things, the fact that yours will become part of the ZMK docs, will force me to think on a different name.

Any plans to have the tool dealing with local build setup? I would be happy to archive my thing if your tool can handle that part.

joelspadin, 2024-08-19:
Setting up local builds isn't currently an area I've been focused on. My primary goal was to make a tool to replace the separate bash and PowerShell setup scripts, but I'm not opposed to adding more subcommands to handle common tasks for local builds.

What particular parts of setting up a local build do you find difficult? Just running a local build that matches the GitHub build?

closed 2025-12-16 by nmunnich