zmkfirmware/zmk #45: Feature: combo behavior

petejohanson, 2020-07-03:
We need to implement a combo behavior, allowing key combos to trigger additional behavior in addition to standard single key presses.

jmding8, 2020-10-04:
I believe that combos, although not hugely popular yet, are very useful and deserve to be featured. The fact that bt_unpair_combo was added is evidence of how useful they are, especially in split-keyboard scenarios. Another example, it is not uncommon for layer selector keys to be on the right-hand keyboard alone. This means that without combos, it is difficult to have a way to trigger &reset or &bootloader without requiring that the right-hand keyboard be working and connected.

okke-formsma, 2020-10-07:
I'm working on this.

Please see the attached pdf that shows some diagrams of how these combos would work. [zmk-combo-proposal.pdf](https://github.com/zmkfirmware/zmk/files/5343404/zmk-combo-proposal.pdf)

tominabox1, 2020-10-08:
> 
> 
> I'm working on this.
> 
> Please see the attached pdf that shows some diagrams of how these combos would work. [zmk-combo-proposal.pdf](https://github.com/zmkfirmware/zmk/files/5343404/zmk-combo-proposal.pdf)

don't see any issues with this proposal, seems to be how they work in QMK as far as I can tell

elrom, 2020-11-17:
Just dropping by to say: +1 to this. I'm anxiously awaiting this feature. The lack of it is currently preventing me from switching from qmk.

okke-formsma, 2020-11-21:
a progress update;

I now have a branch that intercepts key-position-events.
* if a full combo is presssed, the combo is sent
* if a non-combo key is pressed, all intercepted keys are released
* multiple combos can be pressed and released sequentially
* combos can overlap partially (eg. AB and BC, but not AB and ABC)

Not yet working:
* a timeout timer. Current implementation acts as if all timeouts are set to infinite.
* combos are released when all their keys are released instead of when the first key in the combo is released (would be nice to make this configurable)
* fully overlapping combos (e.g. a 3-key-combo overlapping a 2-key-combo)

okke-formsma, 2020-11-22:
Update today; 
* fixed bug that caused normal keys to fail
* fixed bug that caused only combos that were pressed in the specified order to be recognized

todo;
- [ ] timeout timer
- [ ] make chords deactivate on first key release
- [ ] overlapping chords

okke-formsma, 2020-11-22:
some ideas for the future

* [ ] it would be cool if we could set timeouts per chord instead of globally

okke-formsma, 2020-11-24:
note to self; I think combo's dont trigger hold-taps, because they catch the key-events and generate a behavior.

To fix this, we probably need a "behavior event" which the hold-taps can listen to.

tominabox1, 2020-11-24:
> note to self; I think combo's dont trigger hold-taps, because they catch the key-events and generate a behavior.
> 
> To fix this, we probably need a "behavior event" which the hold-taps can listen to.

They do, I have one set up that way now for low budget grave-Esc

okke-formsma, 2020-11-24:
A hold-tap activated by a combo works fine, but a hold-preferred combo does not work well with a combo right now (e.g. alt-tab with alt on a hold-tap and tab as a chord fails currently)

okke-formsma, 2020-11-24:
Update today; adjustable timeout per chord done. The timeout-ms setting is now used.

A trick you can do is define multiple combos on the same key positions. The first one that's not yet timed out will be triggered when you release the keys.

```
		combo_one {
			compatible = "zmk,behavior-combo";
			label = "COMBO_ONE";
			timeout-ms = <100>;
			key-positions = <0 1>;
			bindings = <&kp A>;
		};
		combo_two {
			compatible = "zmk,behavior-combo";
			label = "COMBO_TWO";
			timeout-ms = <200>;
			key-positions = <0 1>;
			bindings = <&kp B>;
		};
		combo_three {
			compatible = "zmk,behavior-combo";
			label = "COMBO_THREE";
			timeout-ms = <300>;
			key-positions = <0 1>;
			bindings = <&kp C>;
		};
```

* press-release within 100ms will trigger combo one and output A
* press-release within 200ms will trigger combo two and output B
* press-release within 300ms will trigger combo three and output C
* press keys longer than 300ms will release keys 0 and 1 and act according to your keymap.

crides, 2020-11-26:
How would different chord behaviours work on different layers? I.e. pressing A & S together will produce 1 on base layer, 2 on raise and 3 on lower. Or would this be supported around when layer-specific chords are supported?

okke-formsma, 2020-11-26:
that's the next step;

```
		combo_three {
			compatible = "zmk,behavior-combo";
			label = "COMBO_THREE";
			timeout-ms = <300>;
			layer = <1>;
			key-positions = <0 1>;
			bindings = <&kp C>;
		};
```

crides, 2020-11-27:
> - combos can overlap partially (eg. AB and BC, but not AB and ABC)

On upstream `okke/chormbos` when there are combos AB -> x and BC -> y, only BC works, outputting y; AB just outputs AB

okke-formsma, 2020-11-28:
Great find @crides. I'll make sure the combos get inserted in the proper order.

okke-formsma, 2020-11-29:
@crides I have fixed partially overlapping combos. Thanks for the great bug report.

tominabox1, 2020-11-29:
Latest push fixed my overlapping combos, thanks @okke-formsma!

okke-formsma, 2020-12-04:
pushed support for overlapping combos.

example: combo AB and ABC defined.  

* AB will trigger if one of these events happens:
  * any key other than C is pressed
  * any key is released
  * AB is held until the timer runs out

okke-formsma, 2020-12-04:
identified a lacking area in the current implementation:

- [ ] when two combos have different timers, a new timer is not started when the first one is processed.

okke-formsma, 2020-12-06:
some more tricky combo scenarios (thanks tom)
* combo AB timeout 50
* combo ABC timeout 100
* AB is pressed within 50ms, C is never pressed.
* expected outcome: AB after 100ms

another one
* combo AB timeout 50
* combo ABCD timeout 100
* AB is pressed within 50ms, C is pressed within 75, D is never pressed.
* expected outcome: AB after 100ms, immediately followed by C.

okke-formsma, 2020-12-06:
The latest commit should address the above issues, but I haven't written all tests for it yet.

okke-formsma, 2020-12-11:
Just pushed some changes that make combos and hold-taps play well together. Once #493 is merged, chords on a specific layer can be implemented.

okke-formsma, 2020-12-16:
I'm running into the following issue. When pressing multiple keys at the same time which are not in a combo, the keys are not properly released.

* have combo's on all keys on the keyboard.
* press `&mt LCTRL A` and `&mt LSHFT B` in a very short interval (less than combo timeout)
* expected: mods lctrl and lshft are pressed
* actual: mod lctrl is pressed, lshft is tapped very shortly aftrer releasing either key.

it seems the second combo's timer never fires, causing the combo to stay in 'waiting' state too long

okke-formsma, 2020-12-20:
The above issue has been fixed by my latest push.

tominabox1, 2020-12-31:
In development of my BLe Chiffre board I found that you can cause ZMK to crash if you accidentally define combos for key positions that don't existing you matrix (i.e. key-positions 30 31 when you only have 28 keys)

There's no indication what is actually happening. it would be nice to fail the compile in the case.

joelspadin, 2020-12-31:
Or if forcing a compile error isn't possible, at least logging an error and not crashing would be the next best thing.

okke-formsma, 2020-12-31:
Forcing a compile error is tricky, but I'll fix the overwriting of unrelated memory and show an error. Good catch!

okke-formsma, 2021-01-01:
added a guard clause and error message to prevent this issue. Thanks @tominabox1

commit feb0d5b9 (2021-01-14): feat(combos): initial implementation

closes #45
closed 2021-01-14 by petejohanson