zmkfirmware/zmk #46: Feature: tap dance

petejohanson, 2020-07-03:
Keymap feature to be implemented: tap dance, allowing different behavior depending on the number of taps of a certain key.

okke-formsma, 2020-12-26:
Tap-dance should not be too crazy to implement;

* we could support any number of taps easily;  see how simple-macro is implemented how that would work.
* for any tap-tap-hold patterns, the last key could be a hold-tap. Thus, tap-dance does not need to concern itself with any tap vs hold decisions.
* Only one tap-dance counter/timer needs to be active at the same time
* tap-dance is interrupted on any position-state-change event

tommyalatalo, 2021-05-11:
Any progress on this feature request?

dxmh, 2021-08-09:
There is ZMK tapdance progress being worked on in this PR: https://github.com/zmkfirmware/zmk/pull/831

WisdomWolf, 2022-01-06:
I'd imagine that the best way to utilize multiple PRs would be to create your own fork and then merge the changes from the repos that have the features that you are interested in.  Then you could use your new repository fork as the source for the west.yaml file.  Shouldn't be too difficult if you are familiar with git, but I haven't tried this myself yet so I could be missing something that makes this more difficult than expected.

kurtis-lew, 2022-04-02:
This issue can be closed, as per PR #1139

closed 2022-04-02 by dxmh