zmkfirmware/zmk #48: Feature: Encoder support

petejohanson, 2020-07-03:
Need to add encoder support that is nicely integrated into ZMK. There are already a couple rotary encoder drivers in Zephyr via the sensor API, but there does not appear to be one for standard "EC11" type ALPS (or clone) encoders.

petejohanson, 2020-07-21:
I have implemented the low level EC11 driver in #59, which I intend to build on for the general encoder/sensor functionality.

petejohanson, 2020-07-22:
Just merged #61 which fully implements this for non-split keyboards. Some remaining work to do for split sensor events going over to the central hub for handling.

jhelvy, 2021-07-01:
Hi, I noticed on the documentation [here](https://zmk.dev/docs/development/new-shield#encoders) that right side sensors aren't yet supported - is that still the case? 

> Note that support for peripheral (right) side sensors over BLE is still in progress.

karnadii, 2021-07-01:
> Hi, I noticed on the documentation [here](https://zmk.dev/docs/development/new-shield#encoders) that right side sensors aren't yet supported - is that still the case?
> 
> > Note that support for peripheral (right) side sensors over BLE is still in progress.

there is a working PR here, you can try it. it is tested work on some split keyboard

jhelvy, 2021-07-01:
Fantastic, PR [#728](https://github.com/zmkfirmware/zmk/pull/728) worked like a charm!

pvinis, 2021-07-07:
I don't know if this is the place to report this, but I think `sensitivity` of encoders would be nice to have. Basically to let us configure every how many "clicks" or every how much "rotation" it should do the action/keypress.

joelspadin, 2021-07-07:
> I don't know if this is the place to report this, but I think `sensitivity` of encoders would be nice to have. Basically to let us configure every how many "clicks" or every how much "rotation" it should do the action/keypress.

I think you're looking for the `resolution` property. (Looks like it's missing from the documentation, though #722 would add it.)

jhelvy, 2021-07-08:
Just noting another solution here for anyone who has an encoder on only the right-side. In that case, you can make the right half central and left half peripheral and your encoder should work. I've done this in my [corne config](https://github.com/jhelvy/zmk-config-corne) and it works. The main changed needed are to adjust your `Kconfig.defconfig` file like this:

```
if SHIELD_JCORNE_LEFT

config ZMK_KEYBOARD_NAME
	default "jCorne"

endif

if SHIELD_JCORNE_RIGHT

config ZMK_KEYBOARD_NAME
	default "jCorne Right"
	
config ZMK_SPLIT_BLE_ROLE_CENTRAL
	default y

endif
```

And in your `keyboard_right.conf` file explicitly note that the right half is central:

```
CONFIG_ZMK_SPLIT_BLE_ROLE_CENTRAL=y
```

closed 2024-08-13 by caksoylar