zmkfirmware/zmk #49: Feature: layer-tap behavior

petejohanson, 2020-07-03:
Keymap feature that needs implementing: layer-tap.

A behavior that binds a key to momentary layer when held, or taps a key when tapped independently.

CrossR, 2020-08-11:
I doubt this would be implemented in a layer specific way, but just as a note, tap vs hold behaviour opens up lots of non-layer related binds as well.

https://beta.docs.qmk.fm/using-qmk/advanced-keycodes/feature_space_cadet is a common feature on smaller keyboards, and is similar to layer tap: Left shift on hold, open paren on tap etc.

okke-formsma, 2020-09-04:
I have an implementation proposal for this in https://github.com/okke-formsma/zmk/tree/modifiers / https://github.com/zmkfirmware/zmk/pull/153

It brings the following changes:

* Modifiers are counted. If you press LSFT twice, only when it's released for the second time the bitflag in the HID is zeroed.
* No more differentiation between `MOD_LSFT` and `LSFT`. It's always `LSFT`.
in my "modifiers" branch, you can still use  `&lsft A x, but `&kp (A | LSFT)` gives the same result.
* This makes mod-taps with multiple mods possible:  `&mt (LSFT | LCTL) A`

The mods are implemented as flags in the third byte of the `keycode`. This overloads the keycodes a bit, but I feel that's warranted in this case.

innovaker, 2020-10-16:
Ummmm, @petejohanson, should this be closed?

petejohanson, 2020-10-16:
Yup!

closed 2020-10-16 by petejohanson