zmkfirmware/zmk #62: Board+Shield SPI definitions are problematic

Nicell, 2020-07-23:
The SPI definition from the nice!nano should be removed because the pins defined there are low drive rated pins by Nordic. This means that you shouldn't run these pins above 10kHz otherwise you may have interference/degraded performance with the antenna. This is somewhat a design flaw of the nice!nano, however, I don't think it's a big deal.

While the Pro Micro pinout defines SPI, no AVR keyboard uses SPI from my understanding. Instead, bit-banging is used. In fact, it seems many keyboards don't even use those SPI pins for bit-banging (which is good news for the nice!nano). They utilize seemingly random pins because they use bit-banging.

Currently the main use of SPI will be addressable RGB lighting (until something else is found). This becomes a bit awkward because of how `led_strip` [(see here)](https://github.com/zephyrproject-rtos/zephyr/blob/master/dts/bindings/led_strip/worldsemi,ws2812-spi.yaml) is defined in the DT. When using SPI, it builds off of an existing SPI definition. However, for each shield, there will most likely be a different pin for RGB. Because of this we can't define inside of the nice!nano or Proton C an SPI output. This compounds with the fact that SPI can't be put on any pin with the Proton C (stm32 architecture limitation), and the nice!nano can't have SPI on any pin without degrading the antenna.

The solution for board + shield duos using SPI peripherals is for the shield to define the SPI. Once again though, we can't assume every pro micro compatible board is capable of having SPI on the pins defined in the shield. Is it possible to make different definitions for each board? If not, shields should probably utilize the bit-banged version of `led_strip`.

Bit-banging will *theoretically* work on any board, but there's still a couple issues. First, the bit-banged version of the WS2812 driver isn't implemented for every architecture. Second is that still you can't bit-bang on low drive rated pins without degrading wireless performance. There still probably needs per-board definitions for shields to adjust the `led_strip` definition.

petejohanson, 2020-07-23:
We have options here.

We certainly can just use the bit-banged version, since that seems to be the norm for now.

You *can" add [per board shield overrides](https://docs.zephyrproject.org/latest/guides/porting/shields.html?highlight=shields#board-specific-shield-configuration) which we could leverage for folks wanting to use SPI bit not using the pro-micro pins.

petejohanson, 2020-07-23:
And yes.... Looking at https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/led_strip/ws2812_gpio.c it looks like we have some platform work to support that.

Nicell, 2020-08-11:
I believe this has been addressed using board specific configurations within shields.

closed 2020-08-11 by Nicell