zmkfirmware/zmk #65: RGB Underglow improvements

Nicell, 2020-07-25:
Here are a list of improvements still to be made to RGB underglow:
- [ ] Add Kconfig options for underglow state defaults
- [ ] Add settings support for underglow state to persist between resets
- [ ] Add Kconfig options to enable/disable each effect to save on storage space
- [ ] Work towards split support of syncing RGB effects and animations
- [ ] Add support for a "layer" effect that can trigger a underglow color if a specific layer is active
- [ ] Add support for using RGB underglow behavior actions through rotary encoders
- [ ] Convert `hsb_to_rgb` to use integers instead of floats to improve performance(?)

coltontcrowe, 2021-08-03:
Are any of these tasks currently being worked on?

Nicell, 2021-08-03:
> Are any of these tasks currently being worked on?

The only one currently being worked on is `Work towards split support of syncing RGB effects and animations` with #547

coltontcrowe, 2021-08-03:
Cool!  I want to try to implement the layer functionality, but I’ll probably convert hsb_to_rgb to integers first since that looks like an easy one.

coltontcrowe, 2021-08-21:
I've been working on converting hsb_to_rgb using integers.  How worried are we if the conversion means numbers may round differently during the conversion process?  I'm noticing a few of the rgb values produced with the function I'm working on are off by 1 compared to the old function.

idesignstuff, 2023-01-02:
How complex would it be to support pwm RGB with this underglow behavior?
I have one RGB led per board on a split (Xiao BLE built-in) and would like to use it to show modes/layers/ble/battery etc.
Currently, because it has a common anode, and each color has its own cathode, they are sort of supported in the backlight, but that can't specify relative brightness, only all of them as a group. 

Currently in my <board>.conf file, I have this:

```
# Uncomment for backlight
CONFIG_ZMK_BACKLIGHT=y

CONFIG_ZMK_BACKLIGHT_AUTO_OFF_IDLE=y
CONFIG_ZMK_BACKLIGHT_BRT_START=50
CONFIG_ZMK_BACKLIGHT_BRT_STEP=10
```

In the keyboard.dtsi file:

```
/ {
    chosen {
        zmk,backlight = &backlight;
    };
};
```

In kconfig.defconfig I have:

```
if ZMK_BACKLIGHT

config PWM
    default y

config LED_PWM
    default y

endif # ZMK_BACKLIGHT
```
In keyboard_left.overlay I have:
```
&pwm0 {
    status = "okay";
    /* Red LED P0_26 or D11, Green LED P0_30 or D12, Blue LED P0_6 or D13 */
    ch0-pin = <26>; /* LED RED */
    ch1-pin = <30>; /* LED GRN */
    ch2-pin = <6>;  /* LED BLU */
}; 
/ {
    backlight: pwmleds {
        compatible = "pwm-leds";
        label = "Backlight LEDs";
        pwm_led_0 {
            pwms = <&pwm0 26>; label = "Backlight LED RED";
        };
        pwm_led_1 {
            pwms = <&pwm0 30>; label = "Backlight LED GRN";
        };
        pwm_led_2 {
            pwms = <&pwm0 6>; label = "Backlight LED BLU";
        };
    };
};  
```

Is it possible to assign separate values for each color with the current state of ZMK?