zmkfirmware/zmk #72: Feature Request: Send mouse scroll events from encoder rotations, instead of page up/down

petejohanson, 2020-08-05:
Sending page up/down on encoder rotation only supports scrolling the actively focused content on most OSes. If we instead were to expose a HID mouse/trackpad, and send vertical scroll events, the experience may be improved for many scenarios.

1. Add an additional HID report to the descriptor for mouse events.
1. Add API to send mouse report updates.
1. Hook those to a new behavior you could bind to the encoder sensor location.

tominabox1, 2020-09-22:
Would this feature also include general implementation of mouse keys?

innovaker, 2020-09-22:
MouseKeys would be a separate feature.

petejohanson, 2020-09-22:
> Would this feature also include general implementation of mouse keys? 

This and mouse keys would both rely on some of the same underlying HID changes needed, but would be implemented slightly differently, likely. 

However, there *is* distinct API for behaviours reacting to key positions versus sensor events, so we could choice to use one behavior for both if we wanted.

el-oso, 2021-12-09:
Hi all, is there some advances on this issue? If there is some test code I would like to try it.

T4CORUN, 2023-01-20:
I was able to leverage what was in [Nick's ](https://github.com/nickconway/zmk/blob/dev/app/dts/behaviors/sensor_rotate_key_press.dtsi) repo: `compatible = "zmk,behavior-sensor-rotate-var"` to make my own behavior that will let me use &mwh and &mmv on a rotary encoder. Got mouse scroll wheel working well on my Rollow's thumb horizontal encoders.  You are more than welcome to review it in my [zmk-config](https://github.com/T4CORUN/zmk-config). 

Happy to try to submit a PR but I have very limited knowledge here and don't want to mess something up

pmalecka, 2024-04-28:
Trying to make this work on the new PR (https://github.com/zmkfirmware/zmk/pull/2027) as well as on top of Nick's repo (linked by @t4corun above), but the scroll wheel binding does not seem to work for me..

Here's the behavior I'm using 
```
scroll_up_down: behavior_sensor_rotate_mouse_wheel_up_down {
	compatible = "zmk,behavior-sensor-rotate";
	#sensor-binding-cells = <0>;
        bindings = <&msc SCRL_UP>, <&msc SCRL_DOWN>;
};
```

then it's defined in the keymap:
```
sensor-bindings = <&scroll_up_down>;
```

I'm obviously missing something, as the scroll works when mapped to a key, but does not work when mapped to the encoder.

caksoylar, 2024-04-28:
You should add a longer `tap-ms` to the sensor rotate behavior, see https://zmk.dev/docs/config/behaviors#sensor-rotation

If that isn't enough to scroll, you might want to increase the max speed of the scroll behavior; e.g. define a larger `ZMK_MOUSE_DEFAULT_SCRL_VAL` before importing or use `MOVE_Y(...)` (see https://github.com/zmkfirmware/zmk/pull/2027/files#diff-96b6a313b7b2d798a719c2c62671b1244ad8a8830f1b716c7612dc4c92462dc3R47)

pmalecka, 2024-04-29:
Thanks for the help! 

Had to both to make it work:
`tap-ms = <20>;`
and 
`#define ZMK_MOUSE_SCRL_VAL 140`

ctranstrum, 2025-02-11:
I think this has been completed in #2477.

Nick-Munnich, 2025-02-11:
It sort of has sort of hasn't I think - the capability *exists*, but in a roundabout manner rather than the direct one that would be desired.

caksoylar, 2025-02-11:
Given encoders use [sensor-rotate behaviors](https://zmk.dev/docs/keymaps/behaviors/sensor-rotate) and they use regular behaviors as bindings and mouse scroll is currently implemented as a regular behavior, I don't see an alternate implementation that is more directly tied to encoders.

We should probably document this as an example though, since the `tap-ms` isn't exposed outside the reference page so it is hard for people to figure out.

Nick-Munnich, 2025-02-11:
I recall some discussion on moving encoders to the input API, which is what I was referring to. I wouldn't be opposed to closing this issue and opening one which asks about moving them to the input API though. Agree on example.

caksoylar, 2025-02-11:
I guess I'll close, as the OP Pete can reopen if he wants to.

closed 2025-02-11 by caksoylar