zmkfirmware/zmk #82: Feature: Leader Key

petejohanson, 2020-08-11:
We should have a leader key behavior that allows you to add custom actions implemented for a certain sequence of keys pressed after the leader key behavior is triggered.

QMK Docs: https://docs.qmk.fm/#/feature_leader_key

I would imagine there would be something like:

```
&leader
```

In your keymap, and you would customize the leader behavior *registration* itself, elsewhere in the keymap, e.g.:

```
&leader {
  sequence_pi {
    keycodes = <P I>;
    action = <&not_sure_what_goes_here>;
  };

  sequence_foo {
    keycodes = <F O>;
    action = <&whatever_it_does>;
};
```

arabshapt, 2022-10-16:
This would be awesome! I am not sure how hard it would be to implement it... I am surprised how few people are wanting this feature.

urob, 2024-12-08:
As noted above there's Nick's PR for a position-based version of this. 

For something closer to Pete's syntax, I created a module at https://github.com/urob/zmk-leader-key. The biggest difference relative to Nick's PR is that it is keycode-based and allows multiple leader instances with distinct sequences.