zmkfirmware/zmk #85: Feature: Grave Escape

Nicell, 2020-08-11:
Grave escape is an extremely common feature used by keyboards without an F-row. We should be able to implement this neatly with a new behavior.

QMK docs for reference: https://beta.docs.qmk.fm/using-qmk/advanced-keycodes/feature_grave_esc

okke-formsma, 2020-09-04:
This feature is partially supported by ZMK already, by using a mod-tap:

`&mt LCTL ESC`

Qmk also replaces the key with backtick ('grave') if a shift or gui is pressed. this is not supported in ZMK yet.

okke-formsma, 2020-12-25:
Two possible approaches:

```
/ {
	behaviors {
		gresc: grave_escape {
			compatible = "zmk,behavior-mod-morph";
			label = "GRAVE_ESCAPE";
			#binding-cells = <0>;
			bindings = <&kp ESC>, <&kp GRAVE>;
            mods = <(MOD_LGUI | MOD_LSFT)>;
            deactivate-mods;
		};
	};
};

usage: &gresc

/ {
	behaviors {
		morph: morph {
			compatible = "zmk,behavior-mod-morph";
			label = "GRAVE_ESCAPE";
			#binding-cells = <2>;
			bindings = <&kp>, <&kp>;
            mods = <(MOD_LGUI | MOD_LSFT)>;
            deactivate-mods;
		};
	};
};

usage: &morph ESC GRAVE
```

This could potentially both be supported by the same behavior implementation.

okke-formsma, 2020-12-25:
The following config would make grave-escape work better in windows & macos, where lgui+` is a shortcut:

```
/ {
    behaviors {
        gresc_shift: gresc_shift {
            compatible = "zmk,behavior-mod-morph";
            label = "GRAVE_ESCAPE";
            #binding-cells = <0>;
            bindings = <&kp ESC>, <&kp GRAVE>;
            mods = <(MOD_LSHIFT)>;
        };
        gresc: grave_escape {
            compatible = "zmk,behavior-mod-morph";
            label = "GRAVE_ESCAPE";
            #binding-cells = <0>;
            bindings = <&gresc_shift>, <&kp GRAVE>;
            mods = <(MOD_LGUI)>;
            deactivate-mods;
        };
    };
};

usage: &gresc
presses ESC normally, LGUI+key = grave, shift + key = ~
```

okke-formsma, 2020-12-26:
I made an implementation of grave-escape like QMK has. The `deactivate-mods` idea discussed above is not yet implemented. If people want that kind of behavior maybe they're better off using layers.

tominabox1, 2020-12-26:
Tested a generalized behavior in my keymap and had the expected behavior
`		ccln: ctrl_colon {
 			compatible = "zmk,behavior-mod-morph";
 			label = "CTRL_COLON";
 			#binding-cells = <0>;
 			bindings = <&kp COMMA>, <&kp SEMI>;
            mods = <(MOD_LCTL|MOD_LSFT)>;
 		};`
macOS Big Sur interpreted all keys as expected. Nice work Okke!

PR #543 (2021-01-23): Implement grave-escape
closes #85

this implementation adds grave-escape, but is implemented in such a way that other "mod-morphs" can be created easily as discussed in the ticket.
commit 5eeb310b (2021-01-23): feat(grave-escape): implement grave-escape

closes #85
closed 2021-01-23 by petejohanson