zmkfirmware/zmk #86: Feature: "Shifted Keycodes",  e.g. assigning `!` to a key press or mod-tap behavior.

petejohanson, 2020-08-12:
Potential fix/idea:
*  Use upper bits of the first parameter to `key_press`, etc to have the enum flag of the mods included. UGH, but not sure how to avoid this if we want this to also work w/ things like mod-tap, where the other parameter is already used.
* Add defines to `keys.h` that are <base keycode> + <held mods>, e.g. something like `#define BANG (NUM_1 + HELD_MODS(MOD_LSFT))`
* Updated "key code state changed" event to include *mods* expected with that keycode.
* Handling in the endpoint logic to inspect the mods and handle accordingly. Need to be careful w/ handling mod state, so we don't *release* mods early, etc.

BrainWart, 2020-08-19:
For people who can't wait for a better solution, brainwart/zmk@e09246a8e664c2e873a21cd8d4903f445156c51d

This does not add new keycodes. It just applies two keys one after another.
Usage: `&mkp LSFT NUM_7`

okke-formsma, 2020-09-04:
I have an implementation proposal for this in https://github.com/okke-formsma/zmk/tree/modifiers / https://github.com/zmkfirmware/zmk/pull/153

It brings the following changes:

* Modifiers are counted. If you press LSFT twice, only when it's released for the second time the bitflag in the HID is zeroed.
* No more differentiation between `MOD_LSFT` and `LSFT`. It's always `LSFT`. You can use  `&lsft A` or `&kp (A | LSFT)` for the same result. A shift key would be `&kp LSFT` (preferred) or `&lsft 0` (bad form but it works)
* This makes mod-taps with multiple mods possible:  `&mt (LSFT | LCTL) A`

The mods are implemented as flags in the third byte of the `keycode`. This overloads the keycodes a bit, but I feel that's warranted in this case.

There has been some discussion about this on discord: https://discordapp.com/channels/719497620560543766/719544230497878116/751160749203652649

okke-formsma, 2020-09-04:
We had a chat on discord about not adding shifted keycodes to the default codes, as they are locale-dependent. (e.g. shifted keycodes result in a different character being typed depending on Dutch/Belgian/French/German/..., UK or US setting.)

By forcing users to use `&lsft NUM_3` or `&kp (NUM_3 | LSFT)`, we make them aware of what is happening.

innovaker, 2020-09-22:
Would this syntax (using MACROs) be more robust in terms of future implementation refactoring?
`LCTRL(LSFT(A))`

It looks uglier, but it could be eased with composite convenience MACROs such as:
`LS()`
`LCS()`
`LSC()`
`LCSA()`
etc.

This would hide some of the implementation detail (bit masks) on primarily user-facing code, potentially easing localized refactors?

tominabox1, 2020-09-22:
I like the idea of concatenating mods like this fwiw

innovaker, 2020-09-22:
This syntax would probably also transfer better into whatever comes after DTS for keymaps/bindings/behaviours, in the future.

okke-formsma, 2020-10-27:
discussed on discord https://discordapp.com/channels/719497620560543766/719544230497878116/770359370814586950

* Top byte will be mod flags.
* Simple macros to add each mod flag to a keycode. One per flag.
* Regular normal keycodes for modifiers will be used for those keys that are "just a shift key"
* Same kp used consistently, no duplicated behavior aliases.

okke-formsma, 2020-10-31:
The modifiers branch now includes "implicit" and "explicit" modifiers. An modifier is "explicit" when it's called like `&kp LEFT_CONTROL`. A modifier is "implicit" when it's added to another key; `&kp LC(A)`.

Explicit modifiers are held until their key is released.

Implicit modifiers are held until another key is pressed. This makes rolling over modified keys possible; pressing `LS(A)` and then `B`, and then releasing `LS(A)` and `B` should output "Ab" (and not AB).

closed 2021-02-19 by okke-formsma