zmkfirmware/zmk #88: User config shields fail to apply a keyboard name to BLE

Nicell, 2020-08-15:
With my nice!60 shield definition in my user config, the BLE name doesn't get applied and just shows a MAC address on devices. (https://github.com/Nicell/zmk-config/tree/master/config/boards/shields/nice60)

USB oddly seems to still have the proper name.

petejohanson, 2020-08-15:
@Nicell can you capture your final `.config` from your build in your action, so we can see what's set at that level? TIA.

Nicell, 2020-08-16:
@petejohanson I've got the .config as another artifact now: https://github.com/Nicell/zmk-config/actions/runs/211180850

Nicell, 2020-08-19:
Taking a closer look, not even `SHIELD_NICE60` is getting defined in the .config. This means that the `def_bool` for `shields_list_contains` is failing in user configs.

PR #111 (2020-08-20): fix(shields): Load Kconfig for out-of-tree shields
* Fixes #88
closed 2020-08-20 by petejohanson