zmkfirmware/zmk #114: Feature: Utilize status LED for different BLE actions

Nicell, 2020-08-21:
Most board replacements have a status LED on them that we can use to notify the user when the board is advertising, when it bonds, and when bonds are cleared. (switching profiles could be a feature in the future too). This would give much better feedback as to what's happening for troubleshooting and just in general.

petejohanson, 2020-08-21:
As I was thinking about this, I thought about some states that would be helpful to indicate:

* No BLE connection
* New BLE connection
* Indicate clearing of all bonds via unpair combo.
* Active connection?

As we noted in Discord (https://discord.com/channels/719497620560543766/719544230497878116/746182170061570109), we might want to have those states be dependent on the selected "profile" or "slot" that is currently selected, to allow cycling between different paired devices, and pairing a new device into a certain slot.

Maybe another status might be:

* Blink # of times for that number slot you just switched to, then *pause*, then do the remaining status blink code?

BrainWart, 2020-08-21:
Would this account for the peripheral pairing too?

petejohanson, 2020-08-21:
@BrainWart We would likely want to "forward" this state over to the peripheral side, but that might be a "phase 2" of this.

petejohanson, 2020-08-21:
(Previous deleted comment was based on me thinking this was a different issue. Oy.)

@BrainWart That's a good question. I think we'd use the *peripheral* LED for that status, so the central side status could be used for the connection status to the host. Seems that would work well.