zmkfirmware/zmk #124: Configured device name is not displayed on macOS

tominabox1, 2020-08-23:
Tested on both Catalina 10.15.6 and Big Sur Public Beta 5. 

Device connects but is not advertised as a keyboard (icon does not indicate keyboard) and is only shown as a MAC address. 

When connected, it works and behaves fine, so this appears to be a minor graphical issue.

Bluetooth Preference Pane:
[image]

System Information bluetooth: 
[image]

luantty2, 2020-08-25:
I have the same issue

Na-Cly, 2020-08-25:
Can confirm and device was having issues pairing. iPhone had no issue with display name or connection.

bendalton, 2020-08-30:
Same experience here

a826889260, 2020-09-21:
The media key cannot be used on the iPad

commit 818f0a1f (2020-10-18): fix(bluetooth): Advertise name + appearance.

* Properly put device name and GAP appearance
  in advertising packets, for proper display in
  macOS, Android, etc.
* Closes #124
closed 2020-10-18 by petejohanson