zmkfirmware/zmk #138: Missing documentation on the user of `ZMK_CONFIG` define for user keymaps, etc.

petejohanson, 2020-08-30:
We only have "implicit" documentation on the use of the `ZMK_CONFIG` variable and corresponding directory.

Docs should include:

* What it is, and what it can contain.
  * Toplevel shield/board config and keymap overrides.
  * `boards/` directory structure for out of tree boards and shields.
* How to build locally using `ZMK_CONFIG`.

petejohanson, 2020-10-09:
@kurtis-lew is working on this for us. Thanks!

kurtis-lew, 2020-10-11:
@petejohanson
I don't remember who it was, I believe it was @chenkevinh, but I think that [this page]( https://zmkfirmware.dev/docs/customization) in the docs serves as a good starting point for further documentation for this issue. I'd think it would be appropriate to add the extra details (motivation behind config directories, what they are/what they contain, how to build locally) here instead of adding an entirely new page. What do you say?

caksoylar, 2022-02-12:
Looks like this was already addressed, closing.

closed 2022-02-12 by caksoylar