zmkfirmware/zmk #152: a

okke-formsma, 2020-09-02:
Based on https://github.com/zmkfirmware/zmk/pull/146#discussion_r481645033 I've been thinking about different ways we can define modifier behaviors. I think we need a few ways to define modifiers for different use cases.

## current situation
We can currently only output a modifier keypress. We can currently do that with the unary `&kp`, but that does not allow combining mods: `&kp LSFT` but no `&kp LSFT | LCTL`. 

##proposal
* Modifier and key press combination in a two-ary behavior: `&mods MOD_LSFT A`. Combine modifiers by `&mods (MOD_LSFT | MOD_LCTL) A`).
* Unary modifier-active behavior: `&sft A`, `&gui A`, `&alt A`, `&rsft A`, `&ralt A`, `&rgui A` and their most common combinations: `&sft_gui A`, `&ctrl_alt A`, `&meh A` ctrl shift alt, `&hypr A` ctrl shift alt gui.

Why go through the trouble of defining the &sft when `&mods MOD_LSFT A` works? It's syntactic sugar when used in most cases, but it also allows keypresses to be bound in things like hold-tap where the first argument is expected to be a unary behavior:
```
		i3: behavior_i3_change_worksapce {
			compatible = "zmk,behavior-hold-tap";
			label = "I3_CHANGE_WORKSPACE";
			#binding-cells = <2>;
			flavor = "tap-preferred";
			tapping_term_ms = <200>;
			bindings = <&gui>, <&gui_alt>;
		};
```
`&i3 1 1` would press gui-1 when tapped and gui-alt-1 when held. 

## implementation
Once the &mods behavior is built, the unary behaviors can be created trivially if we make a "partial behavior application" behavior (aka the Curry behavior):

```
		gui: behavior_gui {
			compatible = "zmk,partial-behavior-application";
			label = "GUI";
			#binding-cells = <1>;
			bindings = <&mods>;
			first_argument = <MOD_GUI>;
		};
```
or maybe like this 
```
		meh: behavior_meh {
			compatible = "zmk,partial-behavior-application";
			label = "MEH";
			#binding-cells = <1>;
			bindings = <&mods (MOD_GUI | MOD_LSFT | MOD_LCTL) PLACEHOLDER>;
		};
```

BrainWart, 2020-09-02:
There has been some talk about this issue on the Discord server.
https://discordapp.com/channels/719497620560543766/719565084208398406/750659299688185898

I've attached part of the conversation
[discord convo.txt](https://github.com/zmkfirmware/zmk/files/5162683/discord.convo.txt)

okke-formsma, 2020-09-02:
summary from discord 

* we don't want a keycode like "HASH" because shift-3 does different things in different locales. If we require users to type `&sft NUM_3`, they are aware of what they are actually inputting. (shift-3)
* If a shifted keycode is held, do we want to have it act like "shift-dn 3-dn shift-up ......... 3up" or "shift-dn 3-dn ........ shift-up 3-up"? If we pick the first, there is no chance of errors because of key rollover, but if you keep the key pressed you get the wrong answer. I'm voting for the second way, so key repeats for ################# work as expected.

okke-formsma, 2020-09-02:
We can do this much simpler than I said in the opening post.

Just define the behaviors like this:
```
		sft_gui: behavior_sft_gui {
			compatible = "zmk,modified-key-press";
			label = "SFT_GUI";
			#binding-cells = <1>;
			modifiers = MOD_GUI | MOD_LSFT | MOD_LCTL;
		}; 
```

if someone needs a combination of modifiers that doesn't exist yet, they can define it themselves as a behavior!

okke-formsma, 2020-09-02:
back to pete's idea in https://github.com/zmkfirmware/zmk/issues/86.

closed 2020-09-02 by okke-formsma