zmkfirmware/zmk #164: Suggestion: Switch copyright headers from "authors per file" to "single authors file"

innovaker, 2020-09-06:
Currently, the copyright header in each file lists each contributor of that file.

I'd like to propose centralizing contributors names into a file at the root - such as `AUTHORS` or `CONTRIBUTORS`  - and using a standard header in code, such as (for instance):
```
/*
 * Copyright (c) 2020 The ZMK Contributors
 *
 * SPDX-License-Identifier: MIT
 */
```
... akin to the current [LICENSE ](https://github.com/zmkfirmware/zmk/blob/main/LICENSE) file.

The "authors per file" vs "single authors file" debate has been around _decades_ but I've noticed that some modern projects (and some companies and foundations) are settling on the latter, namely for these compelling reasons:
* Legally speaking, it's said you don't need the names listed per file for intended copyright to apply (I'm not a lawyer, it's just what I've read!).
* Git provides immutable line-by-line attribution history.
* Duplicate lists of names (beyond a few names) get ridiculous.
* It becomes extra maintenance which often gets overlooked.
* What degree of contribution amounts to copyright-able derived work?  In other words, what merits adding a name?  Editing a typo?  A line?  A function?  Who decides that?  How are disagreements handled?
* It increases the burden (responsibilities) on downstream contributors/projects.
* It's trickier to audit and check automatically.

I personally also find it can be psychologically off-putting for potential contributors - hindering growth - which I call the graffiti effect.  Especially for bug-fixers.

I believe @petejohanson is open to this change of norms, but ultimately it's up to the community and each individual (or company).  Whilst the list file could be created by anyone, only the existing contributors can replace the existing headers.  Mostly that would mean @petejohanson doing a Find+Replace for his solo headers, but some files are already joint derived work (multiple authors) which I'd guess require mutual consent from relevant parties.  The earlier it's done, the easier it would be, which is why I'm raising this suggestion now.

Also, I guess it needn't be a project _policy_, it could just be a norm which anyone could choose to deviate from?

I encourage all contributors - or any interested parties - to chime in with their opinions.  I'd encourage anyone concerned to read around first.  To be clear, I'm not a lawyer - I'm just bringing up the issue for discussion.

Examples background:
* https://www.linuxfoundation.org/blog/2020/01/copyright-notices-in-open-source-software-projects/ (for those who trust the Linux Foundation)
* https://opensource.google/docs/releasing/authors/ (for those who trust Google :) )
* https://opensource.stackexchange.com/questions/8158/copyright-notices-and-multiple-developers (for those who trust strangers :D )
* https://google.github.io/opencasebook/ (for those who enjoy reading case law analysis :O )

innovaker, 2020-09-06:
For future reference, a link to this issue was posted onto Discord (with some replies);
https://discord.com/channels/719497620560543766/719544230497878116/752234276195663882

tominabox1, 2020-09-06:
I really find copyrights in open source sw to be weird. I'm copy pasting 99% of my code, it doesn't seem right to put my name on it like I own it! My 2c

innovaker, 2020-09-07:
Summarising the feedback on Discord ...

nacly
> Given my small contributions idc at all

Nicell
> I'm cool either way
> Benefits are appealing.

CrossR
> I seem to think there was somewhere that they explicitly called out the current way and said they'd need it per file, unless I'm misremembering what I read.
> https://github.com/openmoko/openmoko-usb-oui/pull/15

innovaker
> Good spot CrossR.  @petejohanson, you'd need to clarify their precise requirements (from cuvoodoo?).  My impression from that thread is that the licensing in each header is the critical point.

innovaker, 2020-09-07:
Looking through all the headers, I noticed there's already an MIT file from Intel Corporation:
https://github.com/zmkfirmware/zmk/blob/main/app/drivers/zephyr/ec11.c
I'm guessing this was imported from another project?

This is a good example of files that aren't _native_ to ZMK.  I guess the appropriate course of action for these files is to leave the original header, and only append the ZMK copyright line at the top if changes are made?

innovaker, 2020-09-07:
Comparing existing headers to git history, I think the bulk of contributors have given their views.  Just a few board contributors yet to see it.

@petejohanson, let me know if you need me to follow this up with openmoko on your behalf, otherwise I'll leave it in your hands.

petejohanson, 2020-09-10:
@innovaker Thanks for the leg work on this. The one Intel Corp file was used as skeleton, and then completely hacked to pieces to actually implement that driver, so I'm comfortable replacing that header.

I've opened #172 that should cross this one off. Some of the board contributions never had headers *added*, I will do a second pass of sorting through those, but this should at least fix up our C source code.

commit 68e72f92 (2020-09-10): fix: Switch to single AUTHORS file.* Closes #164
closed 2020-09-10 by petejohanson