zmkfirmware/zmk #179: Add "feature matrix" to intro documentation

petejohanson, 2020-09-13:
The current intro document includes a bulleted list of implemented features, and a second list of missing features. We should instead have one unified feature matrix, with columns for QMK and ZMK, and rows for each feature, with checkboxes in each column as appropriate.

To start with, the current list items in the `intro.md` could he converted, and then we can expand it with more details and items.

Some additional items I can already think of:

- AVR/8-bit support
- mouse keys
- runtime keymap updates (VIA)
- wired split support

innovaker, 2020-10-16:
Thanks a lot @TSquash!  Great work 👍 

I missed the review due to illness, so here are some follow-up suggestions:
- Footnotes might be easier to find if they're directly below the table.
- A simple legend for the symbols - I guess below?
- One Shot Keys (link): https://github.com/zmkfirmware/zmk/issues/160
- Combos (link): https://github.com/zmkfirmware/zmk/issues/45
- Keymaps and Layers (change link): https://zmkfirmware.dev/docs/feature/keymaps
- Perhaps reduce `td` `padding-top` and `padding-bottom` so that more of the table fits into a screen and needs less scrolling?  This would require custom CSS.  Careful not to change it site-wide.

When I get a chance in the future, I'll add my feature/behavior notes/wishlist as well.

innovaker, 2020-11-10:
@Tsquash, did the above reach `main` eventually please?

Tsquash, 2020-11-10:
> @Tsquash, did the above reach `main` eventually please?
I have been working on this on and off for a few days, I am still yet to figure out the row height. At one point I could adjust it, but it would not go smaller than it already is. Let me know how you think I should continue from here.

innovaker, 2020-11-10:
Don't worry, there's no rush, I was just noticed during some issue cleaning.

If you need help figuring out the row stuff, please feel free to contact me on Discord about it, but it's not urgent.

huber-th, 2024-03-22:
I believe this issue can be closed as the feature matrix is in the introduction section of the documentation.

closed 2024-03-22 by caksoylar