zmkfirmware/zmk #181: Update new shield documentation to include split shield info

petejohanson, 2020-09-13:
The current dev docs for adding a new shield does not cover the details on how to add a new split shield. It should be updated to include that info:

- split .conf files
- core `.dtsi` file and then left/right overlays, including gpio settings.
- Kconfig differences

kurtis-lew, 2020-10-10:
@petejohanson I was thinking that for t his issue, the "New Shield" page is getting pretty long. Would it be viable to make a tabbed option for users to select between a split new shield guide or a full-size new shield guide?

kurtis-lew, 2020-10-10:
> @petejohanson I was thinking that for this issue, the "New Shield" page is getting pretty long. Would it be viable to make a tabbed option for users to select between a split new shield guide or a full-size new shield guide?

Actually, scratch this, I have an even better idea of how to format this part of the docs. I'll submit another PR within the next few hours.

petejohanson, 2020-10-10:
> @petejohanson I was thinking that for this issue, the "New Shield" page is getting pretty long. Would it be viable to make a tabbed option for users to select between a split new shield guide or a full-size new shield guide?

Seems like there is enough overlap, you could have one doc, with tabs, and then the common sections don't need to be repeated in two different places, get out of sync, etc.

caksoylar, 2022-02-12:
Looks like this was already addressed, closing.

closed 2022-02-12 by caksoylar