zmkfirmware/zmk #189: docs: Windows Setup documentation: 'west' not found

innovaker, 2020-09-18:
The current setup guide for Windows development has an oversight which I think was introduced by #135.

`pip3 install --user -U west` installs west to the local user
but the suggested `C:\Python38\Scripts` is global

This can trip up unsuspecting users.

Further chat about this can be found here:
https://discordapp.com/channels/719497620560543766/719544230497878116/756592929560199228

innovaker, 2020-10-01:
@petejohanson, I just looked into this deeper.

It looks like the bigger issue here is that Zephyr's official [documentation ](https://docs.zephyrproject.org/2.3.0/getting_started/index.html) suggests ...

Linux: `pip3 install --user -U west`
Windows: `pip3 install west`

whereas ZMK currently suggests `pip3 install --user -U west` for every platform.

Before I approve #190, we should answer this question ... do you want to align ZMK to the Zephyr documentation?

innovaker, 2020-10-02:
There was further discussion about this on [Discord](https://discordapp.com/channels/719497620560543766/719909884769992755/761269469405249537) which may supersede this issue: 
- ZMK installation instructions differ to Zephyr's instructions with respect to installing `west`.
- Zephyr recommends not using user installation for Windows and Mac.
- On Windows, this leads to scripts appearing in `C:\Python38\Scripts\` and being added to the correct PATH by the Python installation.
- This is probably a preferable method.
- ZMK installation instructions should either (re-)align with the ZMK instructions or provide instructions for both scenarios i.e. +/- `user`.

chenkevinh, 2020-10-04:
@petejohanson @innovaker Working on this issue. Will be updating to match Zephyr's instructions and will also link to them.

innovaker, 2020-10-05:
Thanks @chenkevinh! 👍

closed 2020-10-05 by innovaker