zmkfirmware/zmk #191: docs: Windows Setup: source

innovaker, 2020-09-19:
https://zmkfirmware.dev/docs/dev-setup/#environment-variables

Suggests running:  `source zephyr/zephyr-env.cmd`

`source` isn't a command within cmd or Powershell.

innovaker, 2020-09-19:
Please note that this was also flagged by Ehps: https://discordapp.com/channels/719497620560543766/719909884769992755/753516093276160001
but I think it got buried.

chenkevinh, 2020-09-19:
For me when I setup on Windows I found the .cmd didn't work, so I skipped it and just added the GNU ARM Embedded environment variables that we have in the docs right now. Not sure what functionality that may or may not have impacted but I've been able to build boards and shields without any issues so far.

closed 2020-10-05 by innovaker