zmkfirmware/zmk #195: Decide on official mascot

petejohanson, 2020-09-21:
Placeholder issue until I can add all the suggested options for folks to give feedback.

Needed before we can make stickers!

procom32, 2020-09-21:
Zeplin Zebras :)
https://i.imgur.com/V4gY3fy.jpg

Na-Cly, 2020-09-22:
Normally I can only stand to 3 stock fox, but in this case I think Zorro.

petejohanson, 2020-09-29:
[image]

The Zapata Wren seems to be a fairly strong contender, with the Zorro not that far behind. Will consider and make a final decision on this soon. Once that's done, we'll need to find a designer to help out!

petejohanson, 2020-09-30:
Ok, so slept on this for a day. 

Zapata Wren it is! They're super cute, and I think will translate well to both a full mascot, and smaller logo formats.

I'll be closing this issue shortly, and open a new issue to track trying to get the new graphics created.

Thanks everyone!