zmkfirmware/zmk #197: Add installation of `dfu-util` to dev setup guide

petejohanson, 2020-09-21:
To support STM32 based boards, e.g. `proton_c` or `planck`, those are set to use `dfu-util` to flash them. We should update the dev setup/installation guide to include installing `dfu-util` by default. It's a small extra install, that will avoid confusion later for anyone targeting that hardware.

shurikai, 2020-09-21:
You have to download dfu-util from https://github.com/redbear/Duo/blob/master/docs/dfu-util_installation_guide.md and put it somewhere in your path (rename to dfu-util.exe first). This will allow dfu flashing.

petejohanson, 2020-09-21:
On macOS and Linux, you should be able to use your package manager (e.g. `brew`, `dnf`, `apt`) to install `dfu-util` as well.

innovaker, 2020-11-10:
@petejohanson, should this be closed?

closed 2020-11-10 by petejohanson