zmkfirmware/zmk #201: Discord Bot for auto-linking GitHub issues

innovaker, 2020-09-23:
ZMK Discord would benefit from a bot which automatically links to GitHub issues when someone posts a GitHub reference.
i.e. https://docs.github.com/en/github/writing-on-github/autolinked-references-and-urls

Hell, every GitHub project on Discord could probably do with such a bot, but I've yet to see one?  Does one already exist?  If not, can we give back to the wider community?

*Your mission, should you choose to accept it.*  Either:
- Find an existing bot which performs this function.
- Investigate whether ZMK's existing bot(s) can be adapted/improved for this purpose.
- Develop a new bot which performs this function.

*This message will self destruct in 5 seconds.*

Tsquash, 2020-10-11:
Just so everyone knows this cannot be done with dyno.gg or any other discord bot maker I know of. (unless you add each link individually but that isn't very scalable) So if you want to attempt this the best way would most likely be to make it in python.

innovaker, 2020-10-11:
Thanks for looking into that @Tsquash.  I understand that you're not taking it further, so I'll leave the issue open for someone else to pick up.

Wofiel, 2020-10-20:
I thought I'd give this a whirl, what are our requirements for this?

I've whipped up a simple proof of concept that will do issues, pull requests and commits, including multiple per message, but there's no commands to trigger it (could be spammy?) or set defaults per channel yet (as we might want with zmk-studio?).

Works:
- Issues
- PRs
- Commits

Unimplemented features:
- Proper configuration tools (most is hard-coded, secrets/tokens are in environment variables)
- Per-channel default repos?
- Action words?

To be specified:
- Whether the above features are needed
- Repo location
- Where a Python bot might be hosted (I've no problem putting it on a personal server and calling it a donation until it's no longer needed)

[image]

Tsquash, 2020-10-20:
@Wofiel Thanks for working on this! I think making commands for it would be useful, for example, !pr270 or !is67 for PR and issues.   This could definitely be improved that's just what I though of first.  And from what I can think this shouldn't really cause too much spam. For hosting your script there should be a few free hosting options from a quick google search, but I haven't looked very much into them.

innovaker, 2020-10-21:
Wow, nice job @Wofiel!  A lovely surprise to start the day!

My only requirement was to pick up GitHub style references and provide a link - much like you've already prototyped.  A use case being that we often know the ids of ongoing issues off the top of our heads, but when we're posting a quick reply it takes a lot of time to visit GitHub to confirm and create a concise link for others.

I'll add that it should also work on edits because it'll be easy to accidentally write the wrong number when relying on a bot.

> Proper configuration methods (eg. username/repo are hard-coded, secrets/tokens are in environment variables)

Presumably so?  I'm no expert when comes to security.  Perhaps look at how other bots handle it?  Anyone want to weigh in on best practices?

> Cap on number of responses to a single message

This seems wise, just so that it isn't abused.  Perhaps a configurable setting?

> Per-channel default repos?

Yes, this will be useful.

An optional configurable JIRA-esque prefix approach might also be useful in the future.  i.e. `#FW123` and `#S123`.  It's probably best to leave that for now, until (if) the need arises.

> Action words?

I personally think action words aren't necessary if the pattern recognition works fine (much like URLs do currently) and has spam protection.

> Repo location

I guess that's a question of maintenance going forward.  @Wofiel, @petejohanson, any thoughts?

> Where a Python bot might be hosted (I've no problem putting it on a personal server and calling it a donation until it's no longer needed)

Again, I'll ask @petejohanson to chime in on that.

Once again, thanks a lot (to you both!) for your efforts.  This is happening sooner than I anticipated and it really helps lighten the load on contributors.

petejohanson, 2020-10-21:
> I'll add that it should also work on edits because it'll be easy to accidentally write the wrong number when relying on a bot.

Edits are a nice to have, but I wouldn't say a requirement for the first pass at this.

> 
> > Proper configuration methods (eg. username/repo are hard-coded, secrets/tokens are in environment variables)
> 
> Presumably so? I'm no expert when comes to security. Perhaps look at how other bots handle it? Anyone want to weigh in on best practices?

To be clear here, are you saying that the username/repo are hard coded, and the secrets are injected via env. variables with your current code? If so, that should be fine to get us started.
> 
> > Cap on number of responses to a single message
> 
> This seems wise, just so that it isn't abused. Perhaps a configurable setting?

Yeah, Probably good to get this in the first version of this.
> 
> > Per-channel default repos?
> 
> Yes, this will be useful.

Again, a nice to have, if it doesn't make the first cut, we can always evolve it.
> 
> An optional configurable JIRA-esque prefix approach might also be useful in the future. i.e. `#FW123` and `#S123`. It's probably best to leave that for now, until (if) the need arises.

Agreed. Future enhancement as needed.
> 
> > Action words?
> 
> I personally think action words aren't necessary if the pattern recognition works fine (much like URLs do currently) and has spam protection.

Fully agree.
> 
> > Repo location
> 
> I guess that's a question of maintenance going forward. @Wofiel, @petejohanson, any thoughts?

We can make a new repo for this in the zmkfirmware org, when we're ready.
> 
> > Where a Python bot might be hosted (I've no problem putting it on a personal server and calling it a donation until it's no longer needed)
> 
> Again, I'll ask @petejohanson to chime in on that.

Let me ponder this one. Depends on what permissions the bot would have, how comfortable I would be having this on a personal server like that.
> 
> Once again, thanks a lot (to you both!) for your efforts. This is happening sooner than I anticipated and it really helps lighten the load on contributors.

Second that! This looks awesome!

Wofiel, 2020-10-22:
Thanks all for the kind and helpful comments!

I did a little more work, added a basic config system to replace the hard-coded variables and added the per-channel overrides. (see attachments)

>I'll add that it should also work on edits because it'll be easy to accidentally write the wrong number when relying on a bot.
>Edits are a nice to have, but I wouldn't say a requirement for the first pass at this.

Really good idea! I checked it out and edits should be possible, though I'd want to put some timeout on it. I'm not sure if there's an edit time limit in Discord (if there is, it's at least 24 hours in the ZMK Discord)
I would say deletions could be done in a similar way.

Would we want to allow messages that didn't initially invoke the bot respond to edits? Minor point, 

>I personally think action words aren't necessary if the pattern recognition works fine (much like URLs do currently) and has spam protection.

Also tend to agree - the only thing that might mistakenly get picked up at the moment is a 40+ char hash that happened to match a commit in the repo, which should be very unlikely.

>Let me ponder this one. Depends on what permissions the bot would have, how comfortable I would be having this on a personal server like that.

That's more than reasonable - security and verification is always a valid consideration. I believe Discord bots can be restricted based on a set of permissions, but even the basic message sending could be very disruptive if used improperly.

>We can make a new repo for this in the zmkfirmware org, when we're ready.

Sounds good, I'll start writing up the readme side as well.

[image]
[image]

Wofiel, 2020-10-28:
Sorry for the busy week! As an update - I just added a README, so the current state of this has been pushed to this repo: https://github.com/Wofiel/GithubLinkbot

CrossR, 2020-10-28:
You'll probably want to add a license to the new repo, since code without a `LICENSE` isn't unlicensed, more just in a weird no-mans land where people can't tell what license it has.

Wofiel, 2020-10-28:
Great point! Added MIT License in line with ZMK.

joelspadin, 2020-12-08:
I've noticed that sometimes people refer to PRs/issues with something like "PR 42" or "issue 42" instead of "#42". Would be nice to catch those too. Here's a regex that catches all three forms:

`(?:#|issue\s*|pr\s*)(\d+)`

(requires case insensitive flag)

Wofiel, 2021-02-13:
Hey all,

Had some spare hours last night, so added a bunch of stuff that was discussed at the same repo above.

- Watches its last 100 (configurable) responses for edits/deletes
- Uses the new reply functionality for single-embed responses (Discord API limitation)
- Can allow/block posting on all channels by default
- Can allow/block a specified list of channels (eg. only post in #development)

Quick demo in gif form of edits/deletes:
[image]

> I've noticed that sometimes people refer to PRs/issues with something like "PR 42" or "issue 42" instead of "#42". Would be nice to catch those too. 

Sorry I missed this ages ago, December was busy and I missed the notif! I'm a touch apprehensive on "PR 42" or "issue 42", as it seems like it might come up in conversation more regularly, eg. "I've tried merging this PR 3 times" or "I've had this issue 6 times this week". That said, it's probably just as easy for "#1" or "#2" to come up, so that's not a strong case.

innovaker, 2021-03-29:
The bot is now in live service.  Thanks so much @Wofiel! 💯

closed 2021-03-29 by innovaker