zmkfirmware/zmk #202: Implement polling based press detection for matrix KSCAN driver.

petejohanson, 2020-09-23:
Currently, the matrix KSCAN driver in ZMK (`app/drivers/zephyr/kscan_gpio_matrix.c`) *only* implements key press detection via interrupts, which is great for low power.

However, for STM32 wired keyboards, this can be a problem as STM32 doesn't allow enabling interrupts for the same pin on different ports, e.g. `&gpioa 15` and `&gpiob 15`, at the same time.

To support those, we should implement polling based detection for the matrix driver. We *already* have this for the direct wire KSCAN driver (`app/drivers/zephyr/kscan_gpio_direct.c`), so we can use that as a reference for how to do this.

It should be behind a new Kconfig flag, so it can be enabled only for wired builds that aren't going for low power usage.

shurikai, 2020-09-23:
It appears that this behavior is already controlled in the gpio_direct driver using the KConfig flag of `CONFIG_ZMK_KSCAN_GPIO_POLLING=y`. Is there any objection to me reusing this flag for the other driver in this instance?

petejohanson, 2020-09-23:
@shurikai So... I think it may be worthwhile to have a separate config for each. In the case of a funky composite device we may want to turn on polling for one but not the other.

shurikai, 2020-09-23:
Will do.

shurikai, 2020-09-23:
Is there a standard for naming these config options?

petejohanson, 2020-09-23:
@shurikai Somewhat. They should definitely be `ZMK_` prefixed, ideally then followed by the feature/area that will use it, then followed by the specific name of the config. So for this, perhaps something like `ZMK_KSCAN_MATRIX_POLLING` and we rename the existing one to `ZMK_KSCAN_DIRECT_POLLING`.

Thoughts?

shurikai, 2020-09-23:
That seems like a reasonable approach to me. I kept coming up with clumsier names. I’ll make that the name going forward.

closed 2020-10-01 by petejohanson