zmkfirmware/zmk #203: Refactor "user config templates" to be in the main `zmk` repo

petejohanson, 2020-09-23:
Currently the `setup.sh` script (`docs/static/setup.sh`) uses separate git repos:

* https://github.com/zmkfirmware/zmk-config-split-template
* https://github.com/zmkfirmware/zmk-config-template

To clone the template, and then populate the initial user config repo being created by the script. This makes changing `setup.sh` harder, since it requires coordinate of PRs against multiple repos. It also means updates to the templates themselves are PRs against multiple repos, easier to miss some, etc.

We should refactor `setup.sh` to clone just a subdirectory of the main `zmk` repo, e.g. `zmk/templates/split`, `zmk/templates/single`, or `zmk/templates/onboard`, and use that to kickstart the user repo.

We'll need to leverage newer Git functionality for this, e.g.: https://stackoverflow.com/questions/600079/how-do-i-clone-a-subdirectory-only-of-a-git-repository/52269934#52269934

So we should also check the Git version before continuing.

ToDo:

- [ ] Move template files into `templates/*` subdirectories.
- [ ] Add git version check to `setup.sh`
- [ ] Once ready to create the user repo, get the initial baseline files by cloning the appropriate subdirectory of the `zmk` repo.

caksoylar, 2022-04-03:
Related: #1178. I believe this would be addressed by #1183 as well.

nmunnich, 2025-08-14:
No longer relevant, our user config setup has changed significantly since time of writing.

closed 2025-08-14 by nmunnich