zmkfirmware/zmk #205: Add missing headers

innovaker, 2020-09-23:
Some existing files are missing their copyright/licensing header.

The basic template for *.c and *.h files is:
```
/*
 * Copyright (c) 2020 The ZMK Contributors
 *
 * SPDX-License-Identifier: MIT
 */
```

Clearing this up is a prerequisite for #38.

DevMashru, 2020-10-01:
Hey, I would like to work on this.

petejohanson, 2020-10-01:
@DevMashru Thanks! I've set you as assigned to the issue. Let me know if you have any questions while you're working on this.

innovaker, 2020-10-01:
Thanks @DevMashru!

I've realised I probably wasn't clear enough in my original task specification.  @petejohanson, do we also want the headers in non-C files done as well (i.e. *every* file) please?

innovaker, 2020-10-01:
@petejohanson let me know in Discord ...
https://discordapp.com/channels/719497620560543766/719544230497878116/761256624492511283
... that it's OK to merge just the `.h`/`.c` files as a first pass.

DevMashru, 2020-10-02:
@innovaker I definitely misunderstood the issue thinking that the header was required only in C files. I am sorry but I am not really sure if I can add this to all the files, at least right now.

innovaker, 2020-10-02:
Yeah, that's my bad.  The `.h` and `.c` files are a good start for this issue.  Thanks for doing it 👍

PR #227 (2020-10-02): Add missing licensing header
Fix #205
commit c68e3d3e (2020-10-02): Add missing licensing header

Fixes #205
closed 2020-10-02 by petejohanson