zmkfirmware/zmk #208: Discuss: proposed docker instructions for development setup

idan, 2020-09-24:
After chatting a bit with @petejohanson yesterday in discord, I tinkered with getting image builds working for local development on my mac using docker. It works, it's way simpler to set up, and I think there might be some wisdom in making it the recommended development setup.

The series of commands to go from nothing to build on MacOS amounts to this:

```bash
# first, install docker
# Either https://docs.docker.com/docker-for-mac/install/
# or
$ brew install docker
# then...
$ docker pull zmkfirmware/zephyr-west-action-arm
$ git clone https://github.com/zmkfirmware/zmk.git
$ cd zmk
$ docker run -w /zmk -v "$PWD:/zmk" zmkfirmware/zephyr-west-action-arm init "-l app"
$ docker run -w /zmk -v "$PWD:/zmk" zmkfirmware/zephyr-west-action-arm update
$ docker run -w /zmk -v "$PWD:/zmk" zmkfirmware/zephyr-west-action-arm update zephyr-export
$ docker run -w /zmk/app -v "$PWD:/zmk" zmkfirmware/zephyr-west-action-arm build "-b nice_nano -- -DSHIELD=sofle_left"
```

I have a .uf2 file sittin' pretty in `zmk/app/build/zephyr` as a result.

### Pros

1. Standardization — the docker image which powers the GH action is the same one used for development. [Dev/prod parity!](https://12factor.net/dev-prod-parity)
2. The development setup instructions ought to be almost identical across Windows/Linux/MacOS.

### Cons / Annoyances discovered so far

- The `west update` step took a hot minute without appearing to have output.
- `west`: four chars `docker run ...`: one billion chars. Maybe a good little shell script to do the invocations is called for.
- I have some nice!nanos, which flash with UF2s. If direct flashing with `west` is called for, I don't know whether that's possible with docker, and whether that's going to be a giant PITA (giving docker the right USB device, etc etc). Even if it works, it might be very different across OSes.
- … Other stuff I haven't encountered yet!

I haven't started actually developing yet. By not installing any of the development dependencies on my machine, it might also hamper my editor's ability to do nice things like symbol autocompletion. Need to kick the tires further to see how that plays out.

WDYT? Happy to roll a docs PR but wanted a short discussion first.

petejohanson, 2020-09-24:
Thanks for the research on this. I wonder how much we could take this and additionally integrate this w/ VS Code "Remote - Container" support: https://code.visualstudio.com/docs/remote/containers-tutorial

idan, 2020-09-24:
@petejohanson Yup, I'll look into it!

idan, 2020-09-24:
See #209, but with the container integration w/ VS Code it's way way simpler. Terminal commands inside the container are simply `west whatever`.

innovaker, 2020-09-26:
Interesting idea @idan!  Thanks for looking into this.  I tried it myself, but settled with WSL2 instead.

This approach is probably superior however - especially with the [Docker Desktop WSL 2 backend](https://docs.docker.com/docker-for-windows/wsl/) which I've yet to try.  The cross-platform aspect is enticing.  I'll compare the two when I get a chance.

How would updates / maintenance be handled?

idan, 2020-09-26:
@innovaker I think updates are "shut down VS Code and make sure the docker container is stopped, then `docker pull docker pull zmkfirmware/zephyr-west-action-arm:latest`. Any additional changes are only in case of some change to the toolchain. I'll try it out though!

mcrosson, 2020-11-25:
I recently deployed docker desktop + vscode following the instructions as of 2020/11/23 and they worked great.

There was one user who followed the instructions on Windows and ran into a permissions issue but it was related to docker being fickle. If the sources are in ```C:\Users\foo\zmk``` or similar path under the home directory it should Just Work ™ on Windows

noclew, 2020-12-05:
> I recently deployed docker desktop + vscode following the instructions as of 2020/11/23 and they worked great.
>
> There was one user who followed the instructions on Windows and ran into a permissions [...]

Where can I find that instruction?

mcrosson, 2020-12-05:
> Where can I find that instruction?

Which instruction? Setting up Docker + VS Code? The setup is in the main docs at https://zmkfirmware.dev/ and the `C:\Users\foo\zmk` bit is in PR 437

noclew, 2020-12-12:
> Which instruction? Setting up Docker + VS Code? The setup is in the main docs at https://zmkfirmware.dev/ and the `C:\Users\foo\zmk` bit is in PR 437

Ah... that instruction... Sorry, I got confused.

dxmh, 2021-09-20:
Related PR that adds a `docker-compose` workflow to the development setup docs: https://github.com/zmkfirmware/zmk/pull/879

Nick-Munnich, 2024-11-02:
Should be closed by #2548

closed 2024-11-02 by Nick-Munnich