zmkfirmware/zmk #215: Not able to send key strokes after disconnect/reconnect without reset

Wofiel, 2020-09-26:
As in the title, after either turning off Bluetooth on the host device, or moving the ZMK device far enough away to disconnect, on reconnecting, the device isn't able to send keystrokes. This is using a split device, I have not verified behaviour on a non-split yet.

Steps to reproduce:

1. Pair/connect your device to the host
2. Turn off the host device's Bluetooth (or move far enough away to disconnect)
3. Turn on the host device's Bluetooth (or move close again)
4. Press a key

Log file here:

```
[00:00:01.122,436] <dbg> zmk.split_central_discovery_func: [ATTRIBUTE] handle 42
[00:00:01.377,410] <dbg> zmk.split_central_discovery_func: [ATTRIBUTE] handle 43
[00:00:01.392,333] <dbg> zmk.split_central_discovery_func: [ATTRIBUTE] handle 45
[00:00:01.392,425] <dbg> zmk.split_central_subscribe: [SUBSCRIBED]
[00:00:02.017,608] <dbg> zmk.unpair_combo_work_handler: Key position 0 not held,                                                                                                                                                                                                                                              skipping unpair combo
[00:00:02.642,425] <dbg> zmk.split_central_connected: Connected: 14:f6:d8:65:74:                                                                                                                                                                                                                                             1b (public)
[00:00:02.642,456] <dbg> zmk.split_central_process_connection: Current security                                                                                                                                                                                                                                              for connection: 1
[00:00:02.643,737] <dbg> zmk.split_central_process_connection: New connection pa                                                                                                                                                                                                                                             rams: Interval: 12, Latency: 30, PHY: 1
[00:00:02.643,859] <dbg> zmk.connected: Connected <log_strdup alloc failed>
[00:00:03.240,203] <dbg> zmk.security_changed: Security changed: <log_strdup all                                                                                                                                                                                                                                             oc failed> level 2

// Key press

[00:00:15.469,116] <dbg> zmk.kscan_gpio_read_0: Sending event at 3,0 state on
[00:00:15.469,177] <dbg> zmk.zmk_kscan_process_msgq: Row: 3, col: 0, position: 38, pressed: true

[00:00:15.469,207] <dbg> zmk.position_state_changed_listener: 38 bubble (no undecided hold_tap active)
[00:00:15.469,207] <dbg> zmk.zmk_keymap_apply_position_state: layer: 0 position: 38, binding name: KEY_PRESS
[00:00:15.469,238] <dbg> zmk.on_keymap_binding_pressed: position 38 usage_page 0x07 keycode 0xe1
[00:00:15.469,238] <dbg> zmk.hid_listener_keycode_pressed: usage_page 0x07 keycode 0xe1
[00:00:15.469,268] <dbg> zmk.zmk_endpoints_send_report: usage page 0x07
[00:00:15.469,329] <dbg> zmk.destination_connection: Address pointer 0x20009217
[00:00:15.469,329] <dbg> zmk.zmk_hog_send_keypad_report: Sending to NULL? no
[00:00:15.513,946] <dbg> zmk.kscan_gpio_read_0: Sending event at 3,0 state off
[00:00:15.514,007] <dbg> zmk.zmk_kscan_process_msgq: Row: 3, col: 0, position: 38, pressed: false

[00:00:15.514,007] <dbg> zmk.position_state_changed_listener: 38 bubble (no undecided hold_tap active)
[00:00:15.514,038] <dbg> zmk.zmk_keymap_apply_position_state: layer: 0 position: 38, binding name: KEY_PRESS
[00:00:15.514,038] <dbg> zmk.on_keymap_binding_released: position 38 usage_page 0x07 keycode 0xe1
[00:00:15.514,068] <dbg> zmk.hid_listener_keycode_released: usage_page 0x07 keycode 0xe1
[00:00:15.514,068] <dbg> zmk.zmk_endpoints_send_report: usage page 0x07
[00:00:15.514,129] <dbg> zmk.destination_connection: Address pointer 0x20009217
[00:00:15.514,129] <dbg> zmk.zmk_hog_send_keypad_report: Sending to NULL? no

// Disconnect by turning off Windows Bluetooth

[00:00:23.751,678] <dbg> zmk.split_central_disconnected: Disconnected: 14:f6:d8:65:74:1b (public) (reason 21)
[00:00:23.751,770] <dbg> zmk.disconnected: Disconnected from 14:f6:d8:65:74:1b (public) (reason 0x15)

// Turning on Windows Bluetooth

[00:00:32.463,226] <dbg> zmk.split_central_connected: Connected: 14:f6:d8:65:74:1b (public)
[00:00:32.463,226] <dbg> zmk.split_central_process_connection: Current security for connection: 1
[00:00:32.464,691] <dbg> zmk.split_central_process_connection: New connection params: Interval: 36, Latency: 0, PHY: 1
[00:00:32.464,782] <dbg> zmk.connected: Connected 14:f6:d8:65:74:1b (public)
[00:00:32.805,511] <dbg> zmk.security_changed: Security changed: 14:f6:d8:65:74:1b (public) level 2

// Key press

[00:00:38.905,334] <dbg> zmk.kscan_gpio_read_0: Sending event at 3,0 state on
[00:00:38.905,395] <dbg> zmk.zmk_kscan_process_msgq: Row: 3, col: 0, position: 38, pressed: true

[00:00:38.905,426] <dbg> zmk.position_state_changed_listener: 38 bubble (no undecided hold_tap active)
[00:00:38.905,426] <dbg> zmk.zmk_keymap_apply_position_state: layer: 0 position: 38, binding name: KEY_PRESS
[00:00:38.905,456] <dbg> zmk.on_keymap_binding_pressed: position 38 usage_page 0x07 keycode 0xe1
[00:00:38.905,456] <dbg> zmk.hid_listener_keycode_pressed: usage_page 0x07 keycode 0xe1
[00:00:38.905,487] <dbg> zmk.zmk_endpoints_send_report: usage page 0x07
[00:00:38.905,517] <dbg> zmk.destination_connection: Address pointer 0x20009217
[00:00:38.905,548] <dbg> zmk.zmk_hog_send_keypad_report: Sending to NULL? no
[00:00:38.905,548] <err> zmk: FAILED TO SEND OVER HOG: -57
[00:00:38.984,130] <dbg> zmk.kscan_gpio_read_0: Sending event at 3,0 state off
[00:00:38.984,222] <dbg> zmk.zmk_kscan_process_msgq: Row: 3, col: 0, position: 38, pressed: false

[00:00:38.984,252] <dbg> zmk.position_state_changed_listener: 38 bubble (no undecided hold_tap active)
[00:00:38.984,252] <dbg> zmk.zmk_keymap_apply_position_state: layer: 0 position: 38, binding name: KEY_PRESS
[00:00:38.984,283] <dbg> zmk.on_keymap_binding_released: position 38 usage_page 0x07 keycode 0xe1
[00:00:38.984,283] <dbg> zmk.hid_listener_keycode_released: usage_page 0x07 keycode 0xe1
[00:00:38.984,313] <dbg> zmk.zmk_endpoints_send_report: usage page 0x07
[00:00:38.984,344] <dbg> zmk.destination_connection: Address pointer 0x20009217
[00:00:38.984,375] <dbg> zmk.zmk_hog_send_keypad_report: Sending to NULL? no
[00:00:38.984,375] <err> zmk: FAILED TO SEND OVER HOG: -57
```

petejohanson, 2021-01-17:
Can you please test again with latest ZMK? We've made various BT improvements lately, curious if this is still an issue.

Wofiel, 2021-01-18:
Looks to have been solved for the two test cases, good stuff!

closed 2021-01-18 by Wofiel