zmkfirmware/zmk #221: ZMK crashes on key spam.

Na-Cly, 2020-09-29:
ZMK crashes when keys are pressed too rapidly. 

At minimum 1 reset is required to fix once it's occurred. 

Last time I watched it on the debugger the the failed to send message count hit around 40 and then the debugger stopped outputting lines. 

This is a very repeatable issue, so no issue testing it.

leejunseok, 2020-11-03:
I am seeing this issue as well, but only in the peripheral side of my Corne + nice nanos. I've spammed keys on my left half and could not get it to reproduce on that half at all.

Interesting notes:
- the debugger stops immediately, not even finishing the current line of output
- sometimes a key is stuck held and sometimes not
- the left half continues to work in some cases
- output from keys on the right half seems to slow down before the freeze

bcat, 2020-11-16:
I see stuck keys periodically on the primary (central?) side as well, e.g., G getting stuck down on the left side of my Crkbd.

ant0ncuranz, 2020-11-21:
I am experiencing the same issue (using a non-split keyboard with a nice-nano).
It seems only to happen when connected via BLE, no issues with USB.

wyolland, 2020-11-29:
Can confirm that I get the same issue on a Quefrency rev1 split. 
If I only spam keys on either side of the split, then I cannot reproduce this issue.
As soon as I start to spam keys on both sides, one of the keys on the right side repeats indefinitely until I reset the board.

Slusho64, 2020-12-01:
I can reproduce this issue over and over in a few seconds with my quefrency by spamming both sides, with the effect of a key held on the left side, the right side, or the keyboard doesn't input anything, until reset.

petejohanson, 2020-12-03:
So, some additional notes: I *cannot* reproduce this over USB. I believe our semaphore use there may be avoiding issues, and we don't have the same in place for the GATT notifications.

joelspadin, 2020-12-18:
There was another report on Discord that spinning an encoder too fast also triggers this, but only when wireless.

4pplet, 2020-12-18:
I've also been experiencing this. Only on BT.

commit 4aa78a6f (2021-01-15): fix(split): Use queue/work for peripheral events.

* Avoid corruption by using work to process
  peripheral key position events on the main
  work thread, like local kscan events are.
* Fixes #221
closed 2021-01-15 by petejohanson