zmkfirmware/zmk #223: Upgrade to Zephyr 2.4.0

petejohanson, 2020-09-30:
Zephyr 2.4.0 was just released, which includes lots of fixes for the BLE stack, in addition to other changes.

We should explore updating, perhaps once that release has had time to "settle".

- we currently have two patches on top of 2.3.0 that need evaluation:
  - USB send fix, may be unneeded with the semaphore fixes we added.
  - some LD fixes for our native posix tests. I attempted to upstream these, but I think they were reverted. Will need further investigation.

We also have some interesting CMake usage, we'll need to test if change to Zephyr's build impact ours.

benjizhai, 2020-10-12:
https://github.com/benjizhai/zmk/tree/zephyr-2.4.0

USB mode works but somehow bluetooth does not (connection dropping 0x08). Need to explore further.

HW: nrfmicro_13 on lily58

petejohanson, 2020-11-10:
@benjizhai Thanks for doing some legwork on this! I would like to get a few more things stable and rolled out before we disrupt things w/ the Zephyr upgrade. I will *definitely* be looking at your branch when we get there though.

PR #467 (2020-12-14): Zephyr v2.4.0
Closes #223.

Based on:
- https://docs.zephyrproject.org/latest/releases/release-notes-2.4.html
- https://github.com/zephyrproject-rtos/zephyr/releases/tag/zephyr-v2.4.0

There are broad API changes and I've rebased the branch many times over the last two weeks (through at least 2 new features), so it should be scrutinized closely and tested thoroughly.  Gloating rights to whoever finds an oversight or a mistake.
closed 2020-12-14 by petejohanson