zmkfirmware/zmk #225: Create mascot and new logo graphics

petejohanson, 2020-09-30:
In #195, we voted on and have selected the Zapata Wren as our new mascot, which now want to incorporate into both a new mascot image, and also into at logo, in a few formats if possible.

Ideally:
* Available in vector and easter format
* The logo should be scalable or have multiple version to work for website, avatars like on Twitter, and places like Discord
* Can be used to produce stickers

petejohanson, 2020-10-01:
@tominabox1 referred me to `@meatsmoothies#7488` on Discord, who's done various MK things, including deskmat and novelties designs for several sets.

She's expressed interested in contributing to this after/in between her commissioned work and other projects. Thankful for her offer, and looking forward to seeing what she comes up with whenever she has time.

AlexGipp, 2020-10-03:
Hey, i would like to give this one a try.
Where should i put the logo?

petejohanson, 2020-10-03:
> Hey, i would like to give this one a try.
> Where should i put the logo? 

I have approached a designer about this already...
 You're welcome to work on some designs, but I can't promise it would be used at this point...

AlexGipp, 2020-10-03:
Eh thats fine to me, im just gonna make a few birds and send them to you.

AlexGipp, 2020-10-03:
[image]
Hey, could you tell me what you think about these?

Spaceman, 2020-10-05:
The bottom right one is very good in my opinion

innovaker, 2020-12-19:
bomtarnes [posted](https://discord.com/channels/719497620560543766/720014518432038922/789868474905133067) some concept ideas on Discord

keyboard-magpie, 2020-12-19:
[image]
[image]
[image]
[image]

it me

tominabox1, 2020-12-19:
these look too Twitter related imo

tominabox1, 2020-12-19:
2nd to last one looks pretty awesome Bom

keyboard-magpie, 2020-12-19:
> 2nd to last one looks pretty awesome Bom

its the one i'm happiest with

okke-formsma, 2020-12-21:
I love the style and execution but I feel the bird is in a weird pose, possibly due to the angle the reference picture was taken from.

[image]

keyboard-magpie, 2020-12-21:
I can give it a go. The tail up style was more specific to the zapata wren- it seems that's fairly distinctive for it

Nicell, 2021-01-07:
Since there's a lack of movement here, I suggest we make a 15 to 30 day guaranteed contest on [Freelancer.com](https://www.freelancer.com/). This will hopefully attract many designers, which should get many unique designs that can be rated and iterated on.

If we do go down this route, during the contest we'll want to stay active giving feedback to make sure we get the most out of the contest. @petejohanson I could definitely help give feedback (although I'm not quite sure how that works with account permissions).

keyboard-magpie, 2021-01-07:
Dat money tho

pdaoust, 2023-12-12:
> I can give it a go. The tail up style was more specific to the zapata wren- it seems that's fairly distinctive for it

Nobody asked me, but I can confirm that the tail-up style is what distinguishes wrens (besides their noisiness, messy nests, outgoing behaviour, and tiny size). Except [NZ rock wrens](https://en.wikipedia.org/wiki/New_Zealand_rock_wren), who have no tails. Anyhow, if you're looking for a distinctive logo that catches the eye, the unusual way wrens hold their tails is certainly that.