zmkfirmware/zmk #232: Feature: Unicode

yannickjmt, 2020-10-02:
Goal: Allow input of Unicode characters directly from the keyboard.
https://beta.docs.qmk.fm/using-qmk/software-features/feature_unicode for reference

As this is OS dependant, it will require a current_OS parameter that could ideally be controlled by the keyboard (&os WIN to switch to windows, &os LNX for linux, ...)

MacOS: add Unicode Hex Input in System Preferences > Keyboard > Input Sources. LAlt(Unicode_hex_code) for output.
Linux: Already enabled by default. Ctrl(Shift(u+Unicode_hex_code))+space for output.
Windows: Requires Wincompose program. RAlt(u+Unicode_hex_code)+enter for output.

A mapping of unicode hex codes should be available in the keymap.
The second code would be the shifted key
Ex: 
#define EAIG &uc 00E9 00C9 // é É
#define NTIL &uc 00F1 00D1 // ñ Ñ
#define PI &uc 03C0 // π

then add them in a layer with &kp EACC

Nice to have: override by config OS dependant unicode output sequences.
Allow output of chain of chars for stuff like (͡ ° ͜ʖ ͡ °) etc..

innovaker, 2020-10-02:
Thanks @yannickjmt.  We hadn't had enough time to properly spec this yet, so thank you for getting it kick started.  I suggest you search the [Discord ](https://zmkfirmware.dev/community/discord/invite) history for `unicode` for any background discussions.

Your proposed approach is intriguing.  A few initial questions if I may:
- If I'm understanding correctly, the modifiers are the only OS dependent bit?
- Do you envisage developing standard sets of unicode preprocessor defines or requiring users to define their own as/when they need them?
- Were you looking to do this development yourself?
   If so, please be sure to read the developer [docs](https://zmkfirmware.dev/docs/), especially the [Clean Room](https://zmkfirmware.dev/docs/dev-clean-room) policy.

This probably needs to be considered in combination with #177 which is still open for ideas.  To an extent, it probably also depends on a few core issues that are in progress (#21, #86, #153, #213).

I realise QMK already has an approach to this, I'd encourage everyone to think outside the box for ZMK's approach and consider all options first.

For completeness we also had an eye on this discussion https://github.com/kiibohd/KiiConf/issues/30.

yannickjmt, 2020-10-02:
- the OS dependent bit is the sequence or shortcut used to by the different OS. It should be invisible for the user imo. The modifiers should be the same regardless of the OS, the user would only provide the unicodes and the OS.
- I think the users should provide their own unicodes at least at the beginning, but it would be nice to have a list of French, Spanish, Nordic chars later.
- maybe but I don't have a lot of time in the next 2/3 weeks and I am a complete newb in C.

There wasn't much in depth discussion on Discord from what I saw. Yes I pretty took QMK approach and simplified it with my limited understanding...

MvEerd, 2021-04-16:
fwif; You can currently achieve this in a similar way QMK does it by using macros, you'd have to change the 'Compose' key using a define depending on which OS you'd want to do this on
See this example outputting and emoji on Linux
https://github.com/MvEerd/zmk-config/commit/d58bb04218d04a353a9a7ebea80407690c4eec9e

chewxy, 2022-04-28:
Hi, as someone who uses the unicode function in qmk and will be acquiring a keyboard running on zmk, I'd like to share some thoughts. 

## Macros are not great.

Here's why

I recently got back into APL programming. In APL for example `ε` is enlist and `⍷` is find. The normal keyboard way is simply `e e <TAB>` and `E E <TAB>` respectively.  On my current ergodox for example, I simply switch into APL layer, and type ε or press shift and get ⍷. 

With Macros, you'd need to have two layers, one for the standard APL layer, and the other for the shifted APL layer. And that's fraught with errors. 

## You need one more "OS" - emacs

emacs has its own particular way of entering unicode characters (the default keybindings are: `C-x-8 RET`), regardless of OS. If you try the alt method, you'd end up creating 2000+ spaces or newlines in emacs.  See also: https://github.com/qmk/qmk_firmware/pull/16949/

urob, 2022-06-11:
Having build-in unicode support would be great! 

Regarding the previous comment: it is possible to implement shifted unicode versions in a single layer using mod-morph, but doing so is quite cumbersome and not very practical when trying to implement a whole layer of, say, Greek symbols.

For example, to implement the second example from the OP, one could use the following code:

```
macros {
  #define OS_LEAD  &kp RALT &kp U  // OS specific sequence to initialize unicode
  #define OS_TRAIL &kp RET         // OS specific sequence to terminate unicode

  ntil: ntil {
    wait-ms = <5>;
    tap-ms = <5>;
    compatible = "zmk,behavior-macro";
    label = "UTF_NTIL_CAP";
    #binding-cells = <0>;
    bindings = <OS_LEAD &kp 0 &kp 0 &kp D &kp 1 OS_TRAIL>;
  }

  ntil_cap: ntil_cap {
    wait-ms = <5>;
    tap-ms = <5>;
    compatible = "zmk,behavior-macro";
    label = "UTF_NTIL";
    #binding-cells = <0>;
    bindings = <OS_LEAD &kp 0 &kp 0 &kp F &kp 1 OS_TRAIL>;
  }
}

behaviors {
  utf_ntil: utf_ntil {
    compatible = "zmk,behavior-mod-morph";
    label = "UTF_NTIL";
    #binding-cells = <0>;
    bindings = <&ntil>, <&ntil_cap>;
    mods = <(MOD_LSFT|MOD_RSFT)>;
    masked_mods = <(MOD_LSFT|MOD_RSFT)>;  // requires PR #1114 
  };
};

keymap {
   ...
  bindings = < &utf_ntil >;
};

```

urob, 2022-06-23:
For anyone interested, I ended up creating a workaround based on preprocessor macros that lives entirely in the userspace. It's not perfect, but might help until a native solution is available.

This is the main file: https://github.com/urob/zmk-config/blob/master/config/unicode.dtsi.  The file should be sourced directly from the root-node of the keymap file. Once sourced, unicode-keys can be created in two ways:

1. single characters can be created with `UNICODE_LOWER`. For instance, to implement a π-character (03C0) one would call 
     ```
     UNICODE_LOWER(pi,    N0, N3, C, N0)
     ```
    This will create a `&pi` shortcut that can be added to the keymap layout
2. pairs of lower/upper-case characters can be created with `UNICODE_UPPER`. For instance, to implement a ñ/Ñ-key (00F1 and 00D1), one would call
    ```
    UNICODE_UPPER(n_tilde,    N0, N0, F, N1,     N0, N0, D, N1)
    ```
    This will create a `&n_tilde` shortcut that can be added to the keymap layout. It yields "ñ" when pressed without shift, and "Ñ" when shifted.

----
Notes:

- the workaround uses the `masked-mods` PR: https://github.com/zmkfirmware/zmk/pull/1114
- the code is operation-system specific, and requires setting the correct leader/end sequences at the top of the file. (the example in my repo is configured to be used on Windows with WinCompose)
- there is a minor bug with the shifted-sequences when pressing the unicode-shortcut-key *extremely* fast after pressing shift. In this case, the shift key sometimes doesn't get swallowed, resulting in garbage unicode sequences printed to the screen. In practice, I didn't have any issues with that.

razlani, 2022-11-24:
> For anyone interested, I ended up creating a workaround based on preprocessor macros that lives entirely in userspace. It's not perfect, but might help until a native solution is available.
> 
> ~This is the main file: https://github.com/urob/zmk-config/blob/master/config/unicode.dtsi. The file should be sourced near the top of the keymap file (see `base.keymap` in [my configuration](https://github.com/urob/zmk-config) for an example).~
> 
> EDIT: The unicode helpers are now part of my collection of helper macros, available here: https://github.com/urob/zmk-nodefree-config. See the documentation there for details on how to set them up.
> 
> Once sourced, unicode-keys can be created in two ways:
> 
> 1. single characters can be created with `ZMK_UNICODE_SINGLE`. For instance, to implement a π-character (03C0) one would call
>    ```
>    ZMK_UNICODE_SINGLE(pi,   N0, N3, C, N0)
>    ```
>    
>    
>        
>          
>        
>    
>          
>        
>    
>        
>      
>    This will create a `&pi` shortcut that can be added to the keymap layout
> 2. pairs of lower/upper-case characters can be created with `ZMK_UNICODE_PAIR`. For instance, to implement a ñ/Ñ-key (00F1 and 00D1), one would call
>    ```
>    ZMK_UNICODE_PAIR(n_tilde,   N0, N0, F, N1,   N0, N0, D, N1)
>    ```
>    
>    
>        
>          
>        
>    
>          
>        
>    
>        
>      
>    This will create a `&n_tilde` shortcut that can be added to the keymap layout. It yields "ñ" when pressed without shift, and "Ñ" when shifted.
> 
> Final remark: Obviously, this is somewhat hacky and the syntax more cumbersome than a native solution would be. The constraint here is that the preprocessor can't detect types and can't loop. Hence the need to use arguments such as `N0, N3, C, N0` instead of `03C3`. A native solution would hopefully be able to avoid this by "looping" over a single four-character string and prepending numbers by `N` when creating the macro. Otherwise, I think, the implementation here could serve as a blueprint for a behavior. Unfortunately, I am not good enough at C to implement the behavior myself from scratch but would be happy to help out as much as I can if someone is taking on the core work.

Good work herein @urob - will defs be using this for spanish characters whilst a more permanent solution presents itself.

itpropro, 2024-05-13:
Any updates on this? This would really help people who need international characters from their countries layout with a en-us base layout.

zapling, 2024-05-13:
> Any updates on this? This would really help people who need international characters from their countries layout with a en-us base layout.

I'm not using ZMK currently, but to chime in this I think using `compose-key` inside your OS of choice is a better solution in most cases. I relied on sending unicode sequences in the past for swedish chars `åäö` but always had varying success doing this. The result would be dependent on how the application handled the unicode input.

The solution I have now is sending the `compose` key seqences for the character I want, which to me have been way more successful.

Example on how I do it form QMK, which should get you and idea on how to port this over to your layout in ZMK https://github.com/zapling/qmk-atreus62/blob/master/zapling/keymap.c#L58

urob, 2025-08-12:
This is now possible using the [zmk-unicode module](https://github.com/urob/zmk-unicode).

The examples from the original can be implemented using the module as follows:

```c
#include <behaviors/unicode.dtsi>

#define EAIG &uc 0xE9 0xC9 // é É
#define NTIL &uc 0xF1 0xD1 // ñ Ñ
#define PI &uc 0x3C0 0     // π
```

To switch input methods while the keyboard is in use, there are:
- `&uc UC_SET_MACOS`
- `&uc UC_SET_LINUX`
- `&uc UC_SET_LINUX_ALT`
- `&uc UC_SET_WIN_COMPOSE`
- `&uc UC_SET_WIN_ALT`
- `&uc UC_SET_EMACS`

Some of these come with additional configuration options to overwrite the default input characters.

Finally, there are predefined convenience macros for commonly used code points. For instance, the characters from above examples are equivalent to:

```c
&uc UC_ES_E_ACUTE // é É
&uc UC_ES_N_TILDE // ñ Ñ
&uc UC_EL_PI      // π Π
```

wlo2, 2025-10-23:
Great to have as a module but would be really convenient as built-in functionality.

Underdisc, 2026-02-12:
## Is Unicode Really Necessary?
This doesn't seem like zmk's responsibility. zmk simply sends **hid usage ids** [\[spec\]](url) and it's up to the OS what it does with the received hid usage id. For what I assume the majority of use cases are (typing non-English characters), _having access to all of unicode is overkill_. A use case is an _EmojiBlaster 500 Key_, but that's an exception.

#### _*For those looking to add a mixed handful of characters not found bundled together on an existing layout...*_
## Issues With urob's Approach
On Linux it uses a macro that types the LControl+LShift+u+[HEX] Unicode input combo. This works in some programs, but not everywhere. When it does work, there's a small delay and one can very briefly see the macro being entered. This approach will also negate the ability to do repeat on hold.

That being said, I did use this approach alongside WinCompose back when I was daily driving Windows and I recall it working well enough that I didn't complain about it.

## OS/DesktopEnvironment Native Approach
Instead of expecting the keyboard to send a keycode representing the desired character, send a keycode that you're not already using (e.g. F13-F24) and have the OS interpret that keycode as your desired character. This is what I'm doing now and the handful of keys I've added behave just like any other (works everywhere, repeats on hold, etc.).

For Linux users, this page is incredibly valuable for achieving this. https://wiki.archlinux.org/title/Keyboard_input