zmkfirmware/zmk #246: bug(keys):  &kp above 0x65 does not work

innovaker, 2020-10-08:
I'm pretty confident that none of these current `&kp` key codes/names work for *anyone*.

```
#define UNDO 0x7A
#define CUT 0x7B
#define COPY 0x7C
#define PSTE 0x7D

#define CRRT 0xC3
#define PRCT 0xC4
#define LABT 0xC5
#define RABT 0xC6
#define AMPS 0xC7
#define PIPE 0xC9
#define COLN 0xCB
#define HASH 0xCC
#define KSPC 0xCD
#define ATSN 0xCE
#define BANG 0xCF

#define VOLU 0x80
#define VOLD 0x81
```

### Rationale
- ZMK's `hid` system currently guards against (blocks) anything above `GUI 0x65`.
- The NKRO keyboard report currently only has enough space for 13 bytes or 104 codes.
- The default USB HID endpoint max packet size (`CONFIG_HID_INTERRUPT_EP_MPS`) is only 16 bytes which would block anything higher than 0x70.
- Regardless of the above issues, my research suggests that only these codes are likely to have support by an OS:
```
#define UNDO 0x7A
#define CUT 0x7B
#define COPY 0x7C
#define PSTE 0x7D

#define VOLU 0x80
#define VOLD 0x81
```

### Moving Forward
Most - if not all - of these codes are going to be superseded by the keys standardization (#21) and possibly by the upcoming shift/modifiers enhancement (#86).

In the short-term, we need to make it clear to everyone - board/shield makers and users - that ...
- these codes don't currently work
- these key names should not be used in shields etc. (especially incoming PRs)
- some of these key names are about to be deprecated, some superseded (by #21).

### End-to-End Testing
What's a tad surprising is that this isn't a new sets of bugs - they've been around since at least June.  These key codes have been incorporated into user keymaps and these shield keymaps:
- `corne`
- `iris`
- `lily58`
- `sofie`
- `qaz`
 
... which suggests our testing and QA needs improvement.
 
IMO, I think it highlights that we cannot just rely on developer or user feedback.  So I suggest we look into developing verifiable end-to-end testing - from ZMK to OS - which can be run by board/shield developers.  My ad-hoc end-to-end system that I threw together for developing #21 has already paid for itself - timewise - and this issue tops the cake.  I'd suggest we explore the possibilities for this - probably using Docker.

CrossR, 2020-10-08:
I can say that for the Sofle board I added (and possibly others), it was known the keys didn't work at the time but were added since it was assumed they'd be updated to work later. It now looks like with the keys changing names etc, that won't be the case, so perhaps a second pass is needed on some of the boards to remove those keys and swap to the functioning version.

As you've mentioned, changing them probably doesn't make sense until #21 and #86 are closed.

innovaker, 2020-10-08:
Thanks @CrossR.  Yeah, I think we all did.

Agreed, best to wait until those issues are merged before doing a second pass.

okke-formsma, 2020-10-08:
Shall we just remove them from the keys.h file? Break it early.

innovaker, 2020-10-08:
I believe the standing policy is to deprecate over break - where possible.  It's just better from a user-perspective.  Most of them are already set to be relegated to legacy (still there, but undocumented) or superseded - see commits of https://github.com/zmkfirmware/zmk/pull/236 for an example.

okke-formsma, 2020-10-08:
* these keycodes don't work currently
* these keycodes won't work in the future
* any reference to these keycodes in code and keymaps can be copied by new users and frustrate them because they don't work

I would suggest to remove them from the codebase and default keymaps so we don't confuse any new users

innovaker, 2020-10-08:
This was discussed at length on Discord from here onwards:
https://discordapp.com/channels/719497620560543766/719544230497878116/763782056236023819

Unless told otherwise by @petejohanson, we won't be removing or deprecating these codes until #21 is merged.

PR #236 (2020-11-04): feat: HID Usages + Keys Standardisation
- Adds HID Usages for ZMK (#217).
- Standardizes key aliases/codes (#21 - still open for RFC).
- Enhances HID reports to take advantage of the new codes (#234).

### Issues
- Closes #217.
- Closes #21.
- Closes #234.
- Supersedes and closes #246.
- Supersedes and closes #158.
- Supersedes and closes #161.

### Testing
- End-to-end tests covering:
  - all key aliases
  - both the keyboard (keypad) and consumer HID reports
- ZMK on Nordic's nRF52840 Dongle.
  - `kscan-mock` running on dongle systematically pressed and released each key.
- Node scripts on R [... cut]
commit a65295c2 (2020-11-04): feat(keys): Standardize keys

Deprecate superseded names with comments.

Add standardized key names.

Closes #21.  Closes #158.  Closes #161.  Closes #246.
closed 2020-11-04 by petejohanson