zmkfirmware/zmk #271: app/Kconfig is broken

benjizhai, 2020-10-12:
build fails as of latest commit (ba73e5d):
```
-- west build: making build dir /home/benji/git/zmk/app/build pristine
-- west build: generating a build system
-- Using keymap file: /home/benji/git/zmk/app/boards/shields/lily58/lily58.keymap
Including boilerplate (Zephyr base): /home/benji/git/zmk/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/benji/git/zmk/app
-- Zephyr version: 2.3.0 (/home/benji/git/zmk/zephyr)
-- Found Python3: /home/benji/.venv/dev/bin/python3 (found suitable exact version "3.8.5") found components: Interpreter
-- Board: nrfmicro_13
ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found toolchain: zephyr (/home/benji/.local/zephyr-sdk-0.11.4)
-- Found west: /home/benji/.venv/dev/bin/west (found suitable version "0.7.3", minimum required is "0.7.1")
-- Found dtc: /home/benji/.local/zephyr-sdk-0.11.4/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/benji/git/zmk/app/boards/arm/nrfmicro/nrfmicro_13.dts
-- Found devicetree overlay: /home/benji/git/zmk/app/boards/shields/lily58/lily58_left.overlay
-- Found devicetree overlay: /home/benji/git/zmk/app/boards/shields/lily58/lily58.keymap
-- Generated zephyr.dts: /home/benji/git/zmk/app/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/benji/git/zmk/app/build/zephyr/include/generated/devicetree_unfixed.h
Parsing /home/benji/git/zmk/app/Kconfig
Loaded configuration '/home/benji/git/zmk/app/boards/arm/nrfmicro/nrfmicro_13_defconfig'
Merged configuration '/home/benji/git/zmk/app/prj.conf'
Merged configuration '/home/benji/git/zmk/app/boards/shields/lily58/lily58_left.conf'

warning: <choice SYS_PM_POLICY> (defined at /home/benji/git/zmk/app/Kconfig:88) defined with type unknown

warning: <choice SYS_PM_POLICY> (defined at /home/benji/git/zmk/app/Kconfig:88) defined without a prompt

warning: the default selection SYS_PM_POLICY_APP (defined at subsys/power/policy/Kconfig:20) of <choice SYS_PM_POLICY> (defined at /home/benji/git/zmk/app/Kconfig:88) is not contained in the choice

error: Aborting due to Kconfig warnings

CMake Error at /home/benji/git/zmk/zephyr/cmake/kconfig.cmake:217 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /home/benji/git/zmk/zephyr/cmake/app/boilerplate.cmake:511 (include)
  /home/benji/git/zmk/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /home/benji/git/zmk/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:35 (include_boilerplate)
  CMakeLists.txt:18 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -B/home/benji/git/zmk/app/build -S/home/benji/git/zmk/app -GNinja -DBOARD=nrfmicro_13 -DSHIELD=lily58_left
```

commenting out the following lines in app/Kconfig solves the issue.

```
diff --git a/app/Kconfig b/app/Kconfig
index fca4912..3dc39fe 100644
--- a/app/Kconfig
+++ b/app/Kconfig
@@ -85,9 +85,9 @@ if ZMK_SLEEP
 config SYS_POWER_DEEP_SLEEP_STATES
        default y

-choice SYS_PM_POLICY
-       default SYS_PM_POLICY_APP
-endchoice
+#choice SYS_PM_POLICY
+#      default SYS_PM_POLICY_APP
+#endchoice

 config ZMK_IDLE_SLEEP_TIMEOUT
        int "Milliseconds to wait to sleep when going idle"
```

megamind4089, 2020-10-12:
@benjizhai Can you try “west update” once ?

innovaker, 2020-10-12:
Yeah, I believe this is due to an upstream change required for a7496ab06425cab7de5fc7164b4ce5a34dd7107b?  `west update` should clear it up.

innovaker, 2020-10-12:
@benjizhai, please let us know if it works so that we can close the issue.

benjizhai, 2020-10-12:
That did solve the issue indeed.

closed 2020-10-12 by benjizhai