zmkfirmware/zmk #281: Periodic "stuck" keys from peripheral sides of splits

petejohanson, 2020-10-14:


joelspadin, 2020-10-14:
I haven't checked if ZMK supports more than 256 keys, but assuming it does, https://en.wikipedia.org/wiki/Variable-length_quantity would be a good way to keep most reasonable keyboards' key positions down to one byte without limiting the number of keys you support.

innovaker, 2020-10-14:
@joelspadin, it's an interesting idea!  I'm not sure if we'll ever need to support nodes with more than 256 keys tho?

@petejohanson, were you initially thinking of bit flags or an integer array?

It might also need to handle outlier scenarios such as when the final key the user types is the one that gets lost.  That might require a timeout on the central combined with either:
a) requesting an update, or
b) mandating repeat/regular state updates for held keys

It reminds me of some of the strategies discussed in the ZeroMQ guide.

Of course, without nailing down the series of events, I guess we're prodding in the dark.

joelspadin, 2020-10-14:
I would guess that an integer array would result in the smaller average packet size, but bit flags would work well if there's any benefit to having a fixed packet size. Still, either way the packets are going to be tiny unless you're using an integer array and manage to hold down every key on a full-size split keyboard.

Slusho64, 2020-10-16:
I have noticed that the keyboard inputs lag/stutter for a few seconds before this happens. Sometimes both halves of my keyboard will stop inputting at all after some lag/stutter as well. Not clear if the two are related or separate issues. This is on win10 with an Intel bluetooth stack with a big antenna.

leejunseok, 2020-11-03:
There could be overlap with #221, since the symptoms look comparable (keys stuck held indefinitely until reset) and because I've personally found that #221 only happens on the peripheral side, similar to this issue.

Slusho64, 2020-12-01:
While this does usually happen on the peripheral side for me, I have had it happen on the left/primary as well.

meln1k, 2022-01-01:
Another idea: what about attaching a monotonically increasing sequence number to each keypress/release notification, and then do some kind of gap detection in the sequence, and if this happens then ask for the messages to be sent again starting from the last known sequence number?

moergo-sc, 2022-04-23:
Is this issue still current on latest ZMK?

Slusho64, 2022-04-23:
> Is this issue still current on latest ZMK?

For me, the stuck key issue was solved a while ago and the latest big update has solved the lagging as well. Unless others have reported these problems recently, this issue should be ready to close.

closed 2022-05-01 by dxmh