zmkfirmware/zmk #288: Retro tapping: make tap-hold on &mt keys output a held tap action.

krikun98, 2020-10-20:
This feature is vital on 40% boards, where essentially all modifiers are on &mt keys. Currently, ZMK outputs a long hold action, and with, for example, my backspace on (&mt RSFT BKSP) there's no way to hold the key.

okke-formsma, 2020-10-20:
@krikun98 I'm not sure if you are looking for retro tapping or tapping force hold.

From QMK docs: 
https://beta.docs.qmk.fm/using-qmk/software-features/tap_hold#retro-tapping
Retro Tapping: Holding and releasing a dual function key without pressing another key will result in nothing happening. With retro tapping enabled, releasing the key without pressing another will send the original keycode even if it is outside the tapping term.

https://beta.docs.qmk.fm/using-qmk/software-features/tap_hold#tapping-force-hold
Tapping force Hold: When the user holds a key after tapping it, the tapping function is repeated by default, rather than activating the hold function. This allows keeping the ability to auto-repeat the tapping function of a dual-role key. TAPPING_FORCE_HOLD removes that ability to let the user activate the hold function instead, in the case of holding the dual-role key after having tapped it.

krikun98, 2020-10-20:
"Tapping force Hold" is the default mode for ZMK. I didn't know "Retro Tapping" was a pre-existing term, I thought it was something you came up with. I want a possibility to disable "Tapping force Hold", i.e. to mimic the default behavior.
The description for "Retro Tapping" doesn't sound like what I need. I'll rename the issue.

joric, 2021-01-31:
Need this asap for jorne backspace. How to enable autorepeat maybe there's a quick patch I could do? Need that urgently.
My keymap: https://github.com/zmkfirmware/zmk/blob/main/app/boards/shields/jorne/jorne.keymap

okke-formsma, 2021-01-31:
Sorry, no work on this has been done yet. I can't think of a quick fix right now other than disabling the hold tap

okke-formsma, 2021-01-31:
@joric it's your lucky day, if you care to test my branch?

joric, 2021-02-01:
@okke-formsma I didn't get autorepeat with your code. Am I supposed to do a specific tap dance? I have a key `&mt RSHFT BSPC` and I want BSPC to autorepeat on hold.

andreiboyanov, 2021-02-01:
Works nicely for me, thank you so much ! I was waiting for this feature so long. I tested it with my home row mods where I have navigation arrows on a layer and now I'm able to move the cursor quickly with this new thing. Made my Monday :)

All I did to activate it was to add `quick_tap_ms = <200>;` to my home row mods behavior config.

luckyrider, 2021-02-01:
Thank you!
It works `&mt LALT SPC` and `&mt RSHFT BSPC`
But what about `&mt LWR RET` or `&mt RSE TAB` ? It's need on 40% keyboards also

petejohanson, 2021-02-01:
> Thank you!
> It works `&mt LALT SPC` and `&mt RSHFT BSPC`
> But what about `&mt LWR RET` or `&mt RSE TAB` ? It's need on 40% keyboards also

You'd want to update the existing layer tap behavior instance in your keymap, e.g.:

```
&lt { quick_tap_ms = <200>; }
```

Note: this is *not* nested at all under the `/` node, since you're referencing an existing node and adding a property to it.

krikun98, 2021-02-02:
This works amazingly well, thanks!

okke-formsma, 2021-02-02:
@petejohanson should be good now

closed 2021-02-06 by okke-formsma

boosh, 2023-09-17:
This doesn't work on my home row mods. Am I missing something?

```
/ {
    behaviors {
        hm: homerow_mods {
            compatible = "zmk,behavior-hold-tap";
            label = "HOMEROW_MODS";
            #binding-cells = <2>;
            tapping-term-ms = <200>;
            quick-tap-ms = <100>;
            global-quick-tap;
            flavor = "tap-preferred";
            bindings = <&kp>, <&kp>;
        };

        /* Toggle layer on tap, kp on hold */
        kp_tog: behavior_kp_tog {
            compatible = "zmk,behavior-hold-tap";
            label = "kp_tog";
            #binding-cells = <2>;
            flavor = "hold-preferred";
            tapping-term-ms = <200>;
            bindings = <&kp>, <&tog>;
        };
    };
};

&lt {
    quick-tap-ms = <200>;
};
```
With
```
&hm RGUI LEFT_ARROW   &hm LALT DOWN_ARROW   &hm RCTRL UP_ARROW   &hm RSHIFT RIGHT_ARROW
```