zmkfirmware/zmk #299: feat: Develop keymap migration script for standardized keys

innovaker, 2020-10-25:
One of the consequences of #21 is that existing keymaps - both user and default - will subsequently contain deprecated key names.  Although we're planning to gradually phase them out over time, it would be helpful if there was a script to do the grunt work for each developer/user.  The *initial* use case being: user runs script, checks the replacements, adjusts any cosmetics (as necessary), and tests it.  Updates to any contributed keymaps can then be PR-ed with relative confidence.

### Specification
- Must be a multi-platform script written in Python.
- Must replace deprecated key names in a keymap.
- Must not replace or damage anything else!
- Must warn users about any deprecated key names that have no replacement (yet).
- Must accept the filepath of the keymap as an argument.
- May accept a directory path as an argument, under which all keymaps are iteratively replaced (*stretch goal*!)

### Substitutions
This is a list of the current default replacements.  The list's specifics may be updated as #21 goes through its final review cycles, but the general form should stay the same.
```json
[
  { "from": "NUM_1", "to": "N1" },
  { "from": "NUM_2", "to": "N2" },
  { "from": "NUM_3", "to": "N3" },
  { "from": "NUM_4", "to": "N4" },
  { "from": "NUM_5", "to": "N5" },
  { "from": "NUM_6", "to": "N6" },
  { "from": "NUM_7", "to": "N7" },
  { "from": "NUM_8", "to": "N8" },
  { "from": "NUM_9", "to": "N9" },
  { "from": "NUM_0", "to": "N0" },
  { "from": "BKSP", "to": "BSPC" },
  { "from": "SPC", "to": "SPACE" },
  { "from": "EQL", "to": "EQUAL" },
  { "from": "TILD", "to": "NON_US_HASH" },
  { "from": "SCLN", "to": "SEMI" },
  { "from": "QUOT", "to": "QUOTE" },
  { "from": "GRAV", "to": "GRAVE" },
  { "from": "CMMA", "to": "COMMA" },
  { "from": "PRSC", "to": "PSCRN" },
  { "from": "SCLK", "to": "SLCK" },
  { "from": "PAUS", "to": "PAUSE_BREAK" },
  { "from": "PGUP", "to": "PG_UP" },
  { "from": "PGDN", "to": "PG_DN" },
  { "from": "RARW", "to": "RIGHT" },
  { "from": "LARW", "to": "LEFT" },
  { "from": "DARW", "to": "DOWN" },
  { "from": "UARW", "to": "UP" },
  { "from": "KDIV", "to": "KP_DIVIDE" },
  { "from": "KMLT", "to": "KP_MULTIPLY" },
  { "from": "KMIN", "to": "KP_MINUS" },
  { "from": "KPLS", "to": "KP_PLUS" },
  { "from": "UNDO", "to": "K_UNDO" },
  { "from": "CUT", "to": "K_CUT" },
  { "from": "COPY", "to": "K_COPY" },
  { "from": "PSTE", "to": "K_PASTE" },
  { "from": "VOLU", "to": "K_VOL_UP" },
  { "from": "VOLD", "to": "K_VOL_DN" },
  { "from": "CURU", "to": null },
  { "from": "LPRN", "to": "KP_LPAR" },
  { "from": "RPRN", "to": "KP_RPAR" },
  { "from": "LCUR", "to": null },
  { "from": "RCUR", "to": null },
  { "from": "CRRT", "to": null },
  { "from": "PRCT", "to": null },
  { "from": "LABT", "to": null },
  { "from": "RABT", "to": null },
  { "from": "AMPS", "to": null },
  { "from": "PIPE", "to": null },
  { "from": "COLN", "to": null },
  { "from": "HASH", "to": null },
  { "from": "KSPC", "to": null },
  { "from": "ATSN", "to": null },
  { "from": "BANG", "to": null },
  { "from": "LCTL", "to": "LCTRL" },
  { "from": "LSFT", "to": "LSHFT" },
  { "from": "RCTL", "to": "RCTRL" },
  { "from": "RSFT", "to": "RSHFT" },
  { "from": "M_NEXT", "to": "C_NEXT" },
  { "from": "M_PREV", "to": "C_PREV" },
  { "from": "M_STOP", "to": "C_STOP" },
  { "from": "M_EJCT", "to": "C_EJECT" },
  { "from": "M_PLAY", "to": "C_PP" },
  { "from": "M_MUTE", "to": "C_MUTE" },
  { "from": "M_VOLU", "to": "C_VOL_UP" },
  { "from": "M_VOLD", "to": "C_VOL_DN" }
]
```

innovaker, 2020-10-27:
We're currently looking for someone(s) to pick up this task.

If you'd like to kick it off, please leave a comment so that we can open a dialogue and assign it to you.  Thanks!

CrossR, 2020-10-29:
I can do this if nobody else will (and assuming there isn't a dev who only knows Python who wants to contribute to ZMK).

innovaker, 2020-10-29:
Thanks @CrossR!  Nobody else has come forward (so far?).  Please let me know if you start it so that I can assign the issue to you.

BrainWart, 2020-11-04:
~~Looks like the new keycodes are live, @ CrossR. I am of the same mind as you with this issue. It would have been a good opportunity for a Python developer to contribute if they wanted. If you're busy or don't need the load at the time, I think I can handle this as well.~~

There was some discussion about what would be the easiest for all ZMK users as well.
https://discord.com/channels/719497620560543766/719544230497878116/773410057052356618

Seems like a JavaScript app on the website may be most accessible to users who may not have any of the platform tools installed.

innovaker, 2020-11-04:
I only specified Python as it's the most common cross-platform language.  But I'm happy for it to be done in Javascript if there's a path forward there.  Some users don't publish their GitHub repositories so I'd request that it can also be used from node also.

innovaker, 2020-11-04:
I've updated the specification accordingly.

innovaker, 2020-11-04:
I've also updated the mappings.

innovaker, 2020-11-05:
Updated the mappings for `GUI` deprecation.

innovaker, 2020-11-07:
Has anyone started work on this issue please?  The firmware improvements and associated documentation is now in `main`.

joelspadin, 2020-11-08:
I have something that appears to be working on the few keymaps I've tested so far. Will put up something for more testing soon.

PR #344 (2020-12-10): feat(docs): add keymap upgrader
Added a documentation page with a script that upgrades deprecated key codes and behaviors to their replacements.

It does not attempt to make sure that it's only updating things in the bindings arrays, as properly parsing DT files is hard. It just uses regex word boundaries to make sure it's only replacing entire words.

Fixes #299
commit 5aa8a07a (2020-12-10): feat(docs): add keymap upgrader

Added a documentation page with a script that upgrades deprecated key
codes and behaviors to their replacements.

Fixes #299
closed 2020-12-10 by innovaker