zmkfirmware/zmk #314: Add support for using external dongle with split keyboard

xudongzheng, 2020-10-31:


innovaker, 2020-11-05:
The suspense is killing me.

xudongzheng, 2020-11-05:
Most of it has been implemented in `app/src/split/bluetooth/central.c` on https://github.com/ergoblue/zmk/commit/200f1452797c613ba5212a0a20c2f58bedabe516. I wrote that to run alongside my Nordic SDK stuff on the peripherals but should be pretty straightforward to use with a ZMK peripheral with minor changes:

- `static struct bt_uuid *service_uuid = BT_UUID_HIDS;` and `static struct bt_uuid *characteristic_uuid = BT_UUID_HIDS_REPORT;` should reflect the custom service
- removing `position = zmk_matrix_transform_row_column_to_position(position / 14, position % 14);`
- use the current `split_central_eir_found()` and identify by UUID instead of by name.

A couple of things that I think are necessary before a pull request:

- Change to central/peripheral identical logic #318 
- Something's been going wrong on the ZMK peripheral that kind of stops notifying after central disconnects/reconnects a few times. For this reason I haven't been able to actually test with a ZMK peripheral.
- Some way of defining "number of peripherals to try to connect to" and a decision on whether they should be identified by name
- Overall code cleanup

luigi311, 2023-01-08:
Im assuming we will have no bongo cat when using dongle mode based on this comment by @petejohanson about only the centeral side having WPM access or will the logic rework fix that?
https://www.reddit.com/r/ErgoMechKeyboards/comments/osr9zb/comment/i4qjb0j/?utm_source=share&utm_medium=web2x&context=3

xudongzheng, 2023-01-08:
> Im assuming we will have no bongo cat when using dongle mode based on this comment by @petejohanson about only the centeral side having WPM access or will the logic rework fix that?

The pull request just makes it possible to have multiple peripherals instead of just one. Any functionality that is only available on a central will still only be available on a central.

xudongzheng, 2023-09-27:
This is implemented in #836.

closed 2023-09-27 by xudongzheng