zmkfirmware/zmk #315: Nice nano will periodically get stuck in bootloader, requiring bootloader reflash

petejohanson, 2020-10-31:
Since we introduced BLE profiles, we've had many user experience problems w/ their nice nano getting "stuck" in the bootloader, and after reflashing ZMK, the next power cycle it goes directly to bootloader again. Reflashing the bootloader w/ adafruit-nrfutil seems to be the only fix for this.

There seems to be some correlation w/ this issue and heavy BLE profile switching, reported by myself and several other users.

Some guesses:

* We're exhausting flash space for the NVS settings backend. (Explore flash circular buffer?)
* We have timing issues w/ the settings calls, causing settings to write to the wrong location.
* Something else entirely.

Nicell, 2020-10-31:
I have some interesting findings for this issue. I've been able to consistently recreate this issue by spamming cycle effect on RGB, which triggers a read to the settings storage every time. It took somewhere around 600 times for it to break the board.

While playing with the storage partition I noted this:

- Size values below 7000 cause USB and basically the whole application to break
- 7000 when it breaks doesn't go to the bootloader, it goes to a dead application
- 8000 (what it's set to now) when it breaks always goes back to the bootloader

I'm wondering if NVS is improperly erasing flash pages when it fills up. That seems likely. I think we should try a different settings back end to see if the issue is alleviated.

8utl3r, 2020-10-31:
Can confirm this is happening to me. I have a Corne with nice!nano. On the split it seems only to happen to the left half. Let me know if I can help or test a fix. :) that said I have limited understanding of how the firmware build stuff works. As in I'm not sure how to pull changes from the main ZMK repo to my own repo...

petejohanson, 2020-10-31:
> Can confirm this is happening to me. I have a Corne with nice!nano. On the split it seems only to happen to the left half. Let me know if I can help or test a fix. :) that said I have limited [...]

Do you use BT profile switching often?

8utl3r, 2020-10-31:
@petejohanson yup, very frequently

PR #322 (2020-11-03): fix(boards): Remap flash to account for bootloader
Closes #315 🎉 

We lose around 50KB of code space with this change, but I think that's a fine tradeoff!
closed 2020-11-03 by petejohanson