zmkfirmware/zmk #319: Is there any plans to develop mouse key function?

BenzeneKim, 2020-11-01:
Hello, 
I am a student who is planning to make a Bluetooth split mechanical keyboard soon. I think it has advantages in more ways than other open sources, so I'm going to develop it based on ZMK. I'd like to add a mouse function to my keyboard, so I'd like to know if there is a schedule for development of mouse functions.

petejohanson, 2020-11-01:
Definitely plans, but no definitive timeline for when this will be worked on.

innovaker, 2020-11-04:
@BenzeneKim, please talk to me on Discord if you're interested in contributing towards this.  I probably have the best visibility of it currently.

kirilligum, 2023-09-28:
@innovaker 
I'm also interested https://discord.com/channels/877392805654306816/937645688244826154/1156954192527753308 and happy to help. how hard is it? is it just adding the mouse movement code to a few files or something more complicated like writing a driver? qmk has that functionality. 
I just got glove80 and missing the mouse badly :-(

JeffDess, 2023-09-28:
It's been on the works for a while now, but you can definitely run it now if you want with [urob's fork](https://github.com/urob/zmk). I've been using it myself for several months without issues. It does require some tweaking at first to get it right tough, but totally fine to use on your daily driver once you do. It'll sure be nice to have official support someday!

Vaisakhkm2625, 2024-07-03:
> It's been on the works for a while now, but you can definitely run it now if you want with [urob's fork](https://github.com/urob/zmk). I've been using it myself for several months without issues. It [...]

shoud i still use urob's fork? or has anymore work done on it since then?

caksoylar, 2024-07-03:
See #2027.

Vaisakhkm2625, 2024-07-03:
> See #2027.

thanks

nmunnich, 2025-08-04:
We have mouse keys now.

closed 2025-08-04 by nmunnich