zmkfirmware/zmk #347: Documentation: reduced width view of documentation hides the outline with no way to restore

leejunseok, 2020-11-08:
If you open the documentation page (https://zmkfirmware.dev/docs) and reduce the width of your browser to the point where the outline on the left is auto-hidden, there's no way to restore the outline. 

Opening the hamburger menu shows only three options: "Docs", "Blog", and "Github" (the three links that are shown in the top bar when the docs are in full-width display mode.
[image]
[image]

innovaker, 2020-11-08:
Thanks for letting us know.  That's *probably* a bug in Docusaurus or perhaps a setting.

Nicell, 2020-11-08:
This isn't a bug, just bad UX from Docusaurus IMO!

See here: 
[image]

[image]

They move the menu location with very little visual indication.

innovaker, 2020-11-08:
Ha, you're right, and I didn't even spot it.  Thanks @Nicell.

To be fair, we're using Docusaurus v2 which is **alpha**.

Sorry @leejunseok.  I'd suggest you speak to the Docusaurus team about it.

closed 2020-11-09 by innovaker