zmkfirmware/zmk #365: Encoders: Support Push and Tilt Features

mcrosson, 2020-11-11:
Per Discord discussion on 2020/11/11 @ 16:05 EST

When implementing encoder support would it be possible to also include push and tilt features that are present on some encoders.

This is a good example of an encoder that does rotation, push and tilt (up/down/left/right): https://tech.alpsalpine.com/prod/e/html/multicontrol/switch/rkjxt/rkjxt1f42001.html

Support for these kinds of encoders in other firmware like QMK is a major pain as you have to make changes to the matrix scanning code and it's not configurable via any kind of UI. With QMK you must adjust the sources/keymaps directly and re-compile to get the support working properly.

innovaker, 2020-11-11:
@mcrosson, thanks for this!  I've unfortunately struggled to find the Discord discussion.  Any chance you could please copy the Discord link into this issue please (right click on first message → `Copy Message Link`)?

mcrosson, 2020-11-11:
> 
> 
> @mcrosson, thanks for this! I've unfortunately struggled to find the Discord discussion. Any chance you could please copy the Discord link into this issue please (right click on first message → `Copy Message Link`)?

Added link to top of issue text

okke-formsma, 2020-11-24:
[image]

This could be implemented with minimal hardware and clever use of a combo;

* Assuming col2row setup, wire a column to Push and ABCD to rows (with a diode).
* Set up a Combo key for the ABCD keymap positions to to activate the behavior you want on Push.
* For bonus points, wire Com to a row too and set up the Combo key for ABCD+Com. This is not really necessary.

Right now combos are not ready yet, but I expect to finish them up before the end of the month.

closed 2024-08-13 by caksoylar