zmkfirmware/zmk #378: iOS Crashes in Modifier Keys page in Settings App

Wofiel, 2020-11-12:
Steps to reproduce: 

1. Have a ZMK keyboard connected
1. Navigate to Settings->General->Keyboard->Hardware Keyboard->Modifier Keys
1. On tapping the ZMK keyboard at the top of the screen, iOS will become unresponsive immediately, or if not, very soon after, requiring a force restart. (Also has occurred when minimising and re-opening the app to the Modifer Keys screen, and when tapping the Modifier Keys page, but less consistent)

[image]

Environment 

Model: iPad Pro (11-inch)
Model Number: (MTXU2B/A)
Software Version: iPadOS 14.1 (18A8395)

edwinbadillo, 2021-08-04:
Any work around to update the modifier keys in the mean time?

dxmh, 2021-08-04:
Just wanted to clarify this issue happens with bluetooth connections and USB connections.

enbyautumn, 2021-09-19:
Happening for me as well 

- iPhone 8
- iOS 14.4.2
- Jailbroken - Might be able to find some crash logs somewhere?

dxmh, 2021-09-21:
This issue is still happening on the new iPadOS version 15.0 released yesterday (on the latest version of ZMK main).

- Model: iPad Air (4th generation)
- Model number: MYFT2B/A
- Software version: iPadOS 15.0 (19A346)

dxmh, 2021-09-21:
Can anyone confirm if this issue exists for other (non-Apple, non-ZMK) keyboards?

Does the iOS device crash when accessing "Modifier Keys" with only a *QMK* keyboard connected, for example?

dxmh, 2021-09-21:
I found that trying to access this settings screen actually only crashes the iOS device for approximately 9 in 10 attempts! 😅 Every once in a while it’ll lock up for several minutes and then succeed instead of rebooting. 

[image]

I was eventually able to get that Globe key (#947) mapped to `&kp CAPS`.  🎉

dxmh, 2021-10-05:
I believe this issue has now been fixed since https://github.com/zmkfirmware/zmk/pull/726#issuecomment-934012093 was merged to main. 

After re-pairing the keyboard with the following Kconfig setting in `<keyboard>.conf`, I can now access the modifier keys page in the Settings app without issues.

```
CONFIG_ZMK_HID_CONSUMER_REPORT_USAGES_BASIC=y
```

(Either that, or this was fixed by the recent 15.0.1 iPadOS bugfix release).

ayroblu, 2021-11-20:
Thanks for this, was super helpful! Was curious if this is the type of thing that can be default to on so that we don't need people to search for this issue when debugging?