zmkfirmware/zmk #392: Battery level doesn't update on Windows

bcat, 2020-11-16:
My Crkbd showed battery level 98% when initially connected to my Windows desktop, and continues to do the same a day later. My Chrome OS workstation, though, shows battery level 87% and seemingly counting down correctly.

bcat, 2020-11-16:
For comparison, BlueMicro did show a continuously updating battery level on Windows, though the range was weird (fully charged around 50% and dead around 15%).

Wofiel, 2020-11-18:
This definitely occurs for me, but not 100% of the time, so I'm not sure what triggers it. My first guess is that it seems to be determined at pairing time?

Just now, after the battery level being stuck at 40% for about a week the last time I paired, I re-paired one of my Windows devices (Remove device, BT_CLR, re-add) and it's now correctly updating and matching my iOS device.

joelspadin, 2020-12-18:
Mine only seems to update every time it reconnects from deep sleep.

mattyams, 2021-02-26:
I too am having this problem except that after re-pairing to my devices updates to the battery level do not happen. I turned on logging and have attached the log here to reference. 

[https://pastebin.com/i57G0z3z](url)

Brausmith, 2021-11-06:
Same issue for me with a Sofle RGB + Nice!Nano. I am on Windows 11 dev build.
Stuck on 62% which is what it was when first paired.

hyx0329, 2023-03-17:
I encounter a similar issue on my Linux setup. After re-pairing the keyboard with my computer, the problem seems solved.

joelspadin, 2023-11-03:
```
CONFIG_BT_GATT_ENFORCE_SUBSCRIPTION=n
```
in a .conf file provides a way to fix this now.

closed 2023-11-03 by joelspadin