zmkfirmware/zmk #411: Dynamic Numlock State

mcrosson, 2020-11-20:
Would it be possible to have an automatic num lock state based on selected layer? I'm working on a keypad setup that will require the num lock status to be on for certain layers but off for others. It'd be super helpful if users (me) didn't have to turn it on and off explicitly.

okke-formsma, 2020-11-24:
To do this we need a behavior that listens to layer triggers and activates some behavior if a specified layer is active (or rather, if some layer is the top layer). Sounds like a reasonably easy first behavior to write for someone interested.

mcrosson, 2020-12-08:
Closing this issue for the time being, will re-surface if needed *after* the dynamic layer discussions are complete and/or implemented in the core zmk code.

closed 2020-12-08 by mcrosson